Ultimaker/Cura #15173: Cura 5.3 fails to slice a file

pinski1, 2023-04-07:
### Application Version

5.3.0

### Platform

Win 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Load model from https://www.printables.com/model/362768-angle-mount-for-tapo-c200-dekco-dc2l-cameras
2. Orientate so it's flat to the bed
3. Slice

### Actual results

Hits an error and gives up on slicing
No further information is provided

### Expected results

It ought to complete slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This file has worked on 5.2.1
This is the 2nd file I've sliced on 5.3.0, the first threw a warning my minimum fan speed was too slow at 15% even though I've never changed that setting.
[cura.log](https://github.com/Ultimaker/Cura/files/11181799/cura.log)
[UM2_Tapo C200 Angle Mount v6.zip](https://github.com/Ultimaker/Cura/files/11181800/UM2_Tapo.C200.Angle.Mount.v6.zip)

pinski1, 2023-04-07:
Never mind, had tree supports turned on

MariMakes, 2023-05-05:
Hey @pinski1,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm glad your issue was resolved, but it shouldn't have crashed, even with the tree supports turned on 😅 
I took a look at your project file and it seems like a common workaround, moving the object on the buildplate, seems to work in your model.
[image]

I won't close the issue yet, so one of the developers can take a look at your crash. 
Thanks for your report 👍

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker