Ultimaker/Cura #15174: Prints not showing up

Fake1647, 2023-04-08:
### Application Version

5.3.0

### Platform

windows 10

### Printer

Ender v2 pro

### Reproduction steps

1.i opened cura
2.i dropped a file into cura

### Actual results

it says its there but its not showing up

### Expected results

something should have showed up

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

no

nallath, 2023-04-11:
We need a bit more info to help you out. Could you please add the log files?

You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log
Ubuntu/Linux: $USER/.local/share/cura/<Cura version>/cura.log

Fake1647, 2023-04-15:
2023-03-30 22:13:21,688 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1775]: Attempting to read file file:///C:/Users/flame/AppData/Local/Temp/Rar$DRa12704.48406/files/chef_pin.stl
2023-03-30 22:13:21,691 - DEBUG - [JobQueueWorker [9]] UM.Mesh.MeshData.calculateNormalsFromVertices [561]: Calculating normals took 0.0 seconds
2023-03-30 22:13:21,692 - DEBUG - [JobQueueWorker [9]] STLReader.STLReader._read [68]: Loaded a mesh with 768 vertices
2023-03-30 22:13:21,694 - DEBUG - [JobQueueWorker [9]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0009975433349609375 seconds. 64 input vertices. 64 output vertices.
2023-03-30 22:13:21,694 - DEBUG - [JobQueueWorker [9]] UM.FileHandler.ReadFileJob.run [83]: Loading file took 0.0 seconds
2023-03-30 22:13:21,697 - DEBUG - [JobQueueWorker [9]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.000997304916381836 seconds. 64 input vertices. 64 output vertices.
2023-03-30 22:13:21,705 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x0000022CE3E81810>, took 0ms
2023-03-30 22:13:21,722 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=1) TranslateOp.(node=<CuraSceneNode object: 'chef_pin.stl'>,trans.=<200.000,7.410,0.000>), took 0ms
2023-03-30 22:13:21,856 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [creality_ender3]
2023-03-30 22:13:22,073 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [190]: Extruder[creality_base_extruder_0 #2] added to [Creality Ender-3 / Ender-3 v2] at position [0]
2023-03-30 22:13:22,074 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Creality Ender-3 / Ender-3 v2 in list of firmware to check.
2023-03-30 22:13:22,225 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [275]: Creating thumbnail image (just before slice)...
2023-03-30 22:13:22,226 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2023-03-30 22:13:22,227 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.3.0-beta.1\share\cura\resources\shaders\overhang.shader]...
2023-03-30 22:13:22,230 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.3.0-beta.1\share\cura\resources\shaders\transparent_object.shader]...
2023-03-30 22:13:22,231 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.3.0-beta.1\share\cura\resources\shaders\striped.shader]...
2023-03-30 22:13:22,324 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [290]: Starting to slice...
2023-03-30 22:13:22,325 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [309]: Going to slice build plate [0]!
2023-03-30 22:13:22,325 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
2023-03-30 22:13:22,438 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3 took 0.35954976081848145 seconds.
2023-03-30 22:13:22,446 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ChangeAtZ
2023-03-30 22:13:22,448 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: ColorMix
2023-03-30 22:13:22,449 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: CreateThumbnail
2023-03-30 22:13:22,450 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2023-03-30 22:13:22,452 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: DisplayProgressOnLCD
2023-03-30 22:13:22,454 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: FilamentChange
2023-03-30 22:13:22,455 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: InsertAtLayerChange
2023-03-30 22:13:22,457 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: PauseAtHeight
2023-03-30 22:13:22,459 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: RetractContinue
2023-03-30 22:13:22,460 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: SearchAndReplace
2023-03-30 22:13:22,463 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: Stretch
2023-03-30 22:13:22,464 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: TimeLapse
2023-03-30 22:13:22,465 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [206]: Begin loading of script: UsePreviousProbeMeasurements
[... cut ...]
2023-03-30 22:13:45,157 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-03-30 22:13:45.156] [info] Connecting to 127.0.0.1:49674
2023-03-30 22:13:45,162 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'chef_blade.stl'>,trans.=<-2.943,0.000,0.699>), took 2ms
2023-03-30 22:13:45,163 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Connected
2023-03-30 22:13:45,163 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [185]: Backend connected on port 49674

closed 2023-05-03 by github-actions[bot]