Ultimaker/Cura #15175: bordeaux the octopus

redvenom6, 2023-04-08:
### Application Version

5.3

### Platform

windows 10

### Printer

voxelab x2

### Reproduction steps

failed trying to slice

### Actual results

[Bordeaux_The_Octopus_STL (1).zip](https://github.com/Ultimaker/Cura/files/11182995/Bordeaux_The_Octopus_STL.1.zip)

### Expected results

please fix i want to print this

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

wtf

GregValiant, 2023-04-09:
Thanks for the repoort.
I had to move it and rotate it but it did slice.  I didn't find anything wrong with the model.
This bug was a real problem in 5.2 and I guess it hasn't been fixed yet.
Someone from the Cura team will take a look.

redvenom6, 2023-04-09:
Ok thanks I also have a issue with cura 5.3 crashing all the time.

Do you know if this is a common issue at the moment. Also when will someone from cura be looking into it?

GregValiant, 2023-04-10:
There have been a lot of reports lately about Cura crashing.  I'm just the greeter here so I have no idea where the Cura Team is on that but I'm sure it's being looked at.
Since 5.3.0 was just released it may be a bit before a new version comes out and addresses the Crashing bug.
@MariMakes might have some information on that.

redvenom6, 2023-04-10:
Thanks appreciate your time and info

MariMakes, 2023-05-04:
Hey @redvenom6,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

That's a pretty cool model 🤩
I was able to slice it with Cura 5.3.1 on an S7.
It might be related to your settings. Have you tried reverting to the default settings?

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

[image]