Ultimaker/Cura #15176: Cura crashed during Marketplace update

PeterGillham, 2023-04-08:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality CR-10S Pro V2

### Reproduction steps

1. Open Cura 5.3
2. Cura prompts Marketplace is out of date and a Sync is required
3. Synce performed OK 
4. Following acceptance of licences CUra then crahses
5. Cura crashes and forces a reload from backup

### Actual results

Cura crashes prompting a reload from backup

### Expected results

Cura should be OK with MArketplace updates

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No particular project causes this - as not loaded STL by this point

Cura 5.3 was working fine last weekend, there have been some marketplace updates which I have performed.
[cura.log](https://github.com/Ultimaker/Cura/files/11183372/cura.log)

GregValiant, 2023-04-09:
Thanks for the report.
Did this occur after updating Mesh Tools?

PeterGillham, 2023-04-10:
I dont seem to have "Mesh tools " installed - does it go by another name ?

PeterGillham, 2023-04-11:
Sorry _ found it now, yes I beleive it was following a MEsh tools update - just an FYI - 5.2.2 was still installed and that seems to work OK without this issue (but has not been updated)