Ultimaker/Cura #15178: slicing failed with unexpected error

vilmarci, 2023-04-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Flsun Super Racer

### Reproduction steps

Created a new project 
added this model: https://www.printables.com/en/model/275703-flexi-rexi-fully-articulated-t-rex
scaled up to 100%
slice -> error

workaround: converting the file to stl resulted in succesful slicing

### Actual results

Error message: slicing failed with unexpected error

### Expected results

correct slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Flexi-Rexi06mm_0h00m0g.zip](https://github.com/Ultimaker/Cura/files/11183510/Flexi-Rexi06mm_0h00m0g.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11183511/cura.log)

MariMakes, 2023-05-05:
Hey @vilmarci,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I can reproduce your crash. 
This looks like one of the crashes we are working on resolving right now. 

[image]

There are two quick ways to resolve it for now. 
1. Move the model away from the center of the buildplate
2. Turn off support, which your current model doesn't need anyways. 

I hope this helps you get unstuck. I'll add the model to the current test set.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker