Ultimaker/Cura #15179: Parts elevated from build plate are dropped on the plate when sliced

migs-ely, 2023-04-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Load model to build plate.

Select engineering mode - 0.1mm layer height.

raise model any amount from build plate to ensure entire model built on support material ("automatically drop models to the build plate" is deselected in preferences). The model is nicely floating in the air (0.5mm off the build plate). I do this often when I have a model with a tangent touching the build plate to eliminate the horrible flat that is created where the model touches the build plate. 

Model is core 1 (UM Grey TPLA), support is core 2 (UM Breakaway)

Hit Slice.
[UMS5_Chain Link Thingy v4.zip](https://github.com/Ultimaker/Cura/files/11183574/UMS5_Chain.Link.Thingy.v4.zip)

### Actual results

Slice is fine, but the model is dropped onto the build plate. Try this with any amount of Z offset, and every time, it drops it to the build plate to slice. 

[image]

### Expected results

The model stays at the selected Z height and Cura builds support up to the model. See example using Cura 5.2.1 here:

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

It's Miguel ;o)

GregValiant, 2023-04-09:
Thanks for the report.
This is an oddity and it's a Setting Visibility thing, but this is currently the "expected behavior".
The setting you need isn't visible when "Generate Support" is enabled, but it is used when the supports are generated.

- Turn off "Generate Support".
- Go to the Mesh Fixes settings and turn off "Remove Empty First Layers".
- Turn on "Generate Support"

This is a duplicate of at least one previous report (#14989 ).  I'll leave this one open as well.

migs-ely, 2023-04-09:
Hi Greg - thanks for getting back to me. I'm a little confused by your answer - does that mean that in every previous version of Cura this functionality was actually not working in the correct way? Or is it that in 5.3.0 this setting's default condition has been flipped? 

Anyhow - thanks for explaining a workaround!

GregValiant, 2023-04-09:
It has been like this for a couple of versions.  My Cura installs only go back to 4.13.1 but that's the way it was in that version as well. 
It's a definite head scratcher when you are trying to figure out what's going on.

This is from the fdmprinter.def.json file for "remove_empty_first_layers":
"enabled": "**not (support_enable** or support_meshes_present)"

So the behavior of hiding the setting when support is enabled is intentional but since the setting gets used whether it's visible or not I think it should remain visible.  The relationship between "Generate Support" and "Remove Empty First Layers" should also be mentioned in the ToolTips for both settings but it is not.

fieldOfView, 2023-04-11:
What seems to have changed recently is not the behavior of the setting being hidden when supports are active, but that layers that only contain support structures are removed when the option to remove empty layers is enabled.

In previous versions of Cura, no empty layers would be removed when supports are enabled, because these layers would not be empty (support structures are expected to be there).

migs-ely, 2023-04-11:
I agree @fieldOfView - in previous versions, no empty layers would be removed as they were full of support - I often elevate a print off the bed by a few layers if the bottom surface is not flat (and when using breakaway) as it ensures a nice bottom surface without a glossy flat surface where the part touched the glass build plate (UM S5) and then a harsh transition to the supported surface. The functionality in 5.3.0 has definitely changed in this regard.