Ultimaker/Cura #15180: DPI Scaling causes 5.3.0 to load off screen on dual monitor

Kvan-VTC, 2023-04-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Open Cura 5.3.0

### Actual results

Program loads in second monitor but is not visible on either monitor (previois versions loaded with a portion of the window on the first monitor)

### Expected results

That the program would load on the same screen as the loading splash screen

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Changing the DPI settings to override program settings allows cura to open on monitor one in a usable state.

This issue has been present since the release of 5.0, perhaps testing it on a system running dual monitors would help?

MariMakes, 2023-05-05:
Hey @Kvan-VTC,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I use multiple monitors but I am not experiencing what you are describing. 
You are not the only one experiencing this so there are a couple of workarounds in place.

I have a couple of questions to narrow it down.
	1. Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

Kvan-VTC, 2023-05-05:
1. Does Cura start if you disable one of the monitors? - I stipulate "Correctly" as in full view,  Yes 

3. Do you have two monitors placed above each other? Can you try this workaround from the community?  - No

5. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software? - No
6. Do you have a Ryzen Graphics card? Can you try this workaround? - No (Nvidia RTX-3060 16gig )
7. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround?  - No

Dual monitor side by side (monitors identify properly 1/2)
When launching, the launch splash screen appears on monitor 2, the program loads off-screen on the left edge (Just enough of the window showing so it can be dragged larger, then moved on screen on monitor 1.
Changed Override High DPI scaling to "Scaling performed by: System". The splash screen still loads on monitor 2 and program in monitor 1, but it is in full view. 
Windows 10 and 11 have the same behavior.

MariMakes, 2023-05-08:
Hey @Kvan-VTC,

Thanks for the update and the extra information.
That makes me think this might be a duplicate of https://github.com/Ultimaker/Cura/issues/12248

If that's the case, there is something else you can try.
Please try this build and let us know if it resolves your issue.
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

closed 2023-05-29 by github-actions[bot]