Ultimaker/Cura #15182: Cura fails to start, dead processes still listet by task manager

kiniad, 2023-04-08:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Tronxy X5SA 400 2E

### Reproduction steps

Starting Cura 5.3.0 
- The blue window occures with start infos
- The last message is "Loading UI..."
- The blue window disappear
- And nothing more happens

- If you open the task manager, there are 2 dead processes:
    - Slicing engine for cura
    - UltiMaker-Cura.exe

- You can do this again and more dead prcoesses remains in the task manager, but no Cura window anymore.
-
[image]

### Actual results

Nothing, no cura window anymore

### Expected results

window with the slicer cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11183802/cura.log)

fieldOfView, 2023-04-09:
I have (temporarily) removed the MeshTools plugin from the Marketplace, so users are no longer told to update to this problematic release.

See https://github.com/fieldOfView/Cura-MeshTools/issues/36

Sherman-c, 2023-04-11:
Hi 
I just encountered the slimier  issue on Cura 5.3.0, half an hour ago, it was working fine.  I closed the app and open it again ~30mins later, no UI showed up, task manager indicated both UltiMaker-Cura and Slicing engine for Cura were loaded
, but doing nothing.  Any work around available?

P

Sherman-c, 2023-04-12:
Hi Team
I'm not sure it was related or not, after I done the window update, KB3025221, Cura 5.3.0  started up properly without any problem and indicated Mesh Tools 3.7.1 is available and everything seems back to normal.  Thanks Team.

fieldOfView, 2023-04-12:
Since the same issue was reported on Mac OS, I don't think any windows update would change anything with regard to this issue. What I saw is that Cura would *sometimes* open succesfully, and sometimes the window would not show. 

I think after installing the windows update, you were just "lucky" that Cura opened, and you could install the fixed version of MeshTools which has a fix for the issue.

Sherman-c, 2023-04-12:
Hi Aldo,

Thanks for the confirmation, I have updated the Mesh Tools to 3.7.1.
Thanks again.

Regards
Sherman