Ultimaker/Cura #15185: Unable to open an STL file when set the default app to UltiMaker-Cura.exe

onolox, 2023-04-08:
### Application Version

5.3

### Platform

Windows 10

### Printer

Ender S1

### Reproduction steps

In windows 10, when you set to "always open stl files" with UltiMaker-Cura.exe, the app don't open and stays in background. (Set the default app to UltiMaker-Cura.exe)
If you then try to double click on an stl file, close cura, and try to open again. Cura don't open UltiMaker-Cura.exe and CuraEngine.exe stays running on background.

### Actual results

Cura don't open and stays running on background.

### Expected results

Cura opens and shows the stl file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Suporte Lampadas v3.zip](https://github.com/Ultimaker/Cura/files/11184201/Suporte.Lampadas.v3.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/11184202/cura.zip)

TimThein, 2023-04-24:
I am able to launch Cura 5.3.0 with no problem, but I am not able to open an .stl file and have Windows 11 launch Cura, instead Windows asks me to select an app to open the file and after selecting Cura 5.3.0 and clicking `Just once` button, it would ask me again and again and again. I could set Cura as the default app for .stl file under Windows Settings, but Windows still does not launch Cura when I try to open an .stl file.

MariMakes, 2023-05-08:
Hey @onolox,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I asked the team and they confirmed that the best way to associate your STL files with Cura is by downloading the MSI installer instead of the .exe. You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.1

I hope this helps 😇

onolox, 2023-05-08:
Maybe inplement what the .msi does in the .exe? I think it's just a little thing on registry.

closed 2023-05-08 by onolox

ChristophDerndorfer, 2023-05-21:
Seeing the same issue as OP and installing the .msi package after having installed the .exe version doesn't solve it.

onolox, 2023-05-29:
For me it worked, maybe you are pointing to the wrong folder?

ChristophDerndorfer, 2023-07-04:
Solved with the work-around described in #15970 (renaming the exe from UltiMaker-Cura.exe to UltiMakerCura.exe by removing the hypen).