Ultimaker/Cura #15188: 5.3.0 Slicing failed with an unexpected error

vechiato, 2023-04-09:
### Application Version

5.3.0

### Platform

macos ventura 13.3

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

Tried to slice an housing_-_Part_1.stl file from project https://www.thingiverse.com/thing:4941148
Tried to rotate 90 degrees or kept as is when first opened. In both cases slicing failed. 

### Actual results

got a error message: "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

### Expected results

Slicing should work as usual

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3S1PRO_housing_-_Part_1.3mf.zip](https://github.com/Ultimaker/Cura/files/11185142/CE3S1PRO_housing_-_Part_1.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/11185143/cura.log.zip)

GregValiant, 2023-04-10:
Thanks for the report and the files.
I couldn't duplicate this as the model sliced for me on the first try.  I tried but I could not get it to "not slice".
Someone from the Cura team will take a look.

nallath, 2023-04-25:
Thanks for reporting this! We're aware of an issue that causes some models to not slice. We don't have a fix yet, but we are investigating it.

In some cases it can help to move / rotate the model a bit. Although it's not a real fix, you might try this workaround for the moment

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker