Ultimaker/Cura #15189: Crash Report on opening

MikeOxawopper, 2023-04-09:
### Application Version

5.3

### Platform

Windows 10 pro

### Printer

Ender3 v2

### Reproduction steps

1. click cura icon

### Actual results

Fatal error crash report
offers a reset of config files
causes cura to act as if its a new program on a totally new computer
enters a set up wizard
no custom printers
no custom profiles
no custom filaments
everything gone

Copy and paste the back up file cause it to crash again, on the odd try it will open but is very slow.

Custom supports don't work with conical if you click on a model the outline is above it on the screen?
cura was working an hour or so previous and conical was being used when i closed it down

### Expected results

didn't open to load a project?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

nallath, 2023-04-11:
By the look of it it seems to be an issue with a plugin that you have installed. Could you go to your configuration folder, navigate to plugins, and remove that plugin?

fieldOfView, 2023-04-11:
The crashlog indicates the Titlebar Info plugin, but I think the Mesh Tools plugin may be at fault here.
Please read this: https://github.com/fieldOfView/Cura-MeshTools/issues/36