Ultimaker/Cura #15191: Union Overlapping Volumes simply doesn't work

S3D-r, 2023-04-09:
### Application Version

5.2.1

### Platform

Win 10

### Printer

Creality CR-10 Max

### Reproduction steps

Put two bodies on platform, make ém overlap. Select _Union Overlapping Volumes_ .  Select _Remove Mesh Intersection_. Slice. Preview.

### Actual results

There will be an outer wall in the edge of the overlap.

### Expected results

The overlapping volumes are supposed be treated as it was a single body. The correct outcome should be infill.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Overlapping bodies.zip](https://github.com/Ultimaker/Cura/files/11185749/Overlapping.bodies.zip)

GregValiant, 2023-04-10:
Thanks for the report.
I couldn't get it to work as I expected either.
I exported you model from Cura as an STL and then merged the combined model using MS 3D Builder.  As a workaround this looks acceptable.  I don't know how your real models would work out though.
[image]

S3D-r, 2023-04-10:
Yeah, your workaround is commonly known, and sometimes it works, but not always: the workaround may raise other issues.

nallath, 2023-04-11:
This is intended behavior. See the extra info that the setting guide gives:
[image]

GregValiant, 2023-04-12:
@nallath I get this with Overlapping Volumes turned off.
[image]
and this with it turned on.
[image]
And this one with Remove Mesh Intersections enabled
[image]
Am I missing something?  I don't think I've ever gotten models to slice like either one of those in the Setting Guide.

S3D-r, 2023-04-12:
> This is intended behavior. See the extra info that the setting guide gives: [image]

OK... So, Cura is just missing the feature where separate intersecting meshes would be printed as one mesh. That would be useful feature, and many other slicers have it.

GregValiant, 2023-08-14:
This is in 4.13.1.  The models on the left are separate meshes and the model on the right are the same two models but I merged them in MS 3D Builder.  In both of these images "Alternate Mesh Removal" is turned off.
[image]

This is the same project in 5.4.0.
[image]

This time I turned on "Alternate Mesh Removal".
4.13.1
[image]

5.4.0
[image]

With this test both versions produced nearly the same result.  It isn't what I expected though.  With AMR turned on there are skins on every layer of the overlapped models.  I thought that was odd.