Ultimaker/Cura #15193: Monitor Tab not showing anything

Arketsu, 2023-04-09:
### Application Version

5.3.0 

### Platform

Windows 11

### Printer

Creality Ender 3 S1

### Reproduction steps

1. updated to version 5.3.0 monitor tab not showing any buttons, progress or webcam only shows message that says. in order to monitor your print with cura please connect printer.
2. Opened version 5.2.2 no issues at all monitor tab works flawlessly.

More info:
Using octoprint plugin every past version worked fine until 5.3.0

### Actual results

new version wont work only on old version.

### Expected results

monitor tab should work on 5.3.0 but it doesn't.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fieldOfView, 2023-04-09:
Logs would help diagnose what is going on, but since this is an issue with 3rd party plugins, you are probably going to have to open an issue at the respective plugin issue queues:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues
https://github.com/emtrax-ltd/Cura2MoonrakerPlugin/issues

Arketsu, 2023-04-09:
@fieldOfView Just did, thank you.

fieldOfView, 2023-04-10:
Thanks! I'll put on my other hat and look in to it from the OctoPrint plugin side of things.
Because - for now - this looks like an issue with 3rd party plugins, I will close this issue on the Cura repository.

closed 2023-04-10 by fieldOfView