Ultimaker/Cura #15195: Discard Changes does not stay on the same profile.

plduganaman, 2023-04-09:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Taz6

### Reproduction steps

This is a "before" image before pressing the circular arrow.  The profile is currently on "PLA Taz6"  You can see the grey star indicating there are changes. 

[image]

After pressing the Circular arrow is the "After" image and you can see the profile switched to "Normal - 0.15 mm"

[image]

### Actual results

If I click the circular arrow it changes the profile to a default profile which is not the current profile.  The disk icon brings up the window I showed which is missing the discard current changes.  Previously selecting discard current changes would remove any changes to the current profile and not go to a different profile.   The only work around I have is to select a different profile, discard the changes at that point, then move back to the original profile. 

### Expected results

Expected after using the discard changes that the profile is still the same profile and not a different profile

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Hopefully the information above should be enough to reproduce the issue.  If not I will be glad to submit anything else requested.

plduganaman, 2023-04-09:
[image]

Corrected AFTER image

nallath, 2023-04-11:
I think that's indeed a mistake on our side. I will take it up with the rest of the team!

plduganaman, 2023-04-28:
Apparently still not fixed in 5.3.1.

plduganaman, 2023-07-04:
Still not fixed in 5.4.0

MariMakes, 2023-10-23:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about resetting your profile. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11194

Thanks for the report! 👍

plduganaman, 2024-08-02:
I guess this will always go to backlog and never fixed.