Ultimaker/Cura #15196: Cura 5.3.0 dosent open the first time

gidleym, 2023-04-09:
### Application Version

5.3.0

### Platform

Windows 11,  i7-9700, Nvidia GeForce RTX 2060

### Printer

Creality CR10 S Pro

### Reproduction steps

I updated Extensions from Marketplace, Next time I opened Cura it Failed and I did a backup and reset. Now I try to open Cura and it fails the first time but not the second time. I can see two instances running in the task manager. One in Background processes and the Apps.
I have uninstalled it and reinstalled.
[image]

### Actual results

Cura works correctly.

### Expected results

I would expect Cura to work the first time it is executed. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11186510/cura.log)

GregValiant, 2023-04-10:
Thanks for the report.
When you updated from the MarketPlace was "Mesh Tools" one of the updates?  It has been removed from the MarketPlace until it's repaired but some of us downloaded the update and had problems similar to what you are experiencing.  #15182 has more information.  Also see [THIS PAGE](https://github.com/fieldOfView/Cura-MeshTools/issues/36) for additional regarding the Mesh Tools problem.

gidleym, 2023-04-10:
Yes it was. I think it was one of the updates that I did.

gidleym, 2023-04-10:
I just got into Cura5.3 and uninstalled the mesh tools extension and it seemed to work.

fieldOfView, 2023-04-10:
A fixed version is now available for manual installation. I have submitted it to the Marketplace, but it may take some time before it is available there.

https://github.com/fieldOfView/Cura-MeshTools/releases/tag/v3.7.2-DEV
Download the "Cura5.0-current.curapackage " and drop the file onto the Cura buildplate as if you were opening a 3d model. Then restart Cura as instructed.