Ultimaker/Cura #15199: Error slicing model after scale operation

ddesborough, 2023-04-10:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3v2

### Reproduction steps

Loaded a model of NH and sliced it, it was fine. Scaled the model by 260% and sliced again. Then I received "Slicing failed with an unexpected error. Please consider reporting a bug", so I am. The file in question is NH.stl from here: https://www.thingiverse.com/thing:5256690/files

[CE3E3V2_NH.zip](https://github.com/Ultimaker/Cura/files/11187319/CE3E3V2_NH.zip)

### Actual results

Received an Unexpected error. 
[image]

Log file:
[cura.zip](https://github.com/Ultimaker/Cura/files/11187322/cura.zip)

### Expected results

It should slice successfully

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

All included above

MariMakes, 2023-05-05:
Hey @ddesborough,

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.3 Stable.
Could you test if that's the case for you as well? 

[image]

You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

ddesborough, 2023-05-08:
I have finally had a chance to go back and test this. It does appear that the error has been fixed. I tested with 5.3.1. 

Just to be sure it wasn't due to the state of my machine at the time of the report, I loaded the 3MF in 5.2.1 again and could still reproduce the issue there. 

So, we are all good now. Thanks!

closed 2023-05-08 by ddesborough