Ultimaker/Cura #15201: Older versions of CURA I use 4.4.1 & 4.8, when I opened stated the install process all over again? Had & used for years!

middlechamber, 2023-04-10:
### Application Version

4.4.1, 4.8

### Platform

Win 10/11

### Printer

3- Creality Ender 3 Pros

### Reproduction steps

I had already been using 4.4.1 printing for a couple of hours on Friday. I had just removed a print & went back to the PC to configure/print a new file, when I opened the "file link" the app all of a sudden opened the installation of the program as if I had never used it in the 4+yrs. Not knowing what was going on I had a print deadline to make so I then opened 4.8 another version I use and after the program opened to the default screen that too decided it had to be installed even though it was installed when it was first issued. 
So I went to the last version I use 5.2.3 Christmas Alpha & that worked no problem w/o it going into a reinstall.

### Actual results

After it finished the install all of my settings were gone in both 4.4 & 4.8.  
I noticed in the beginning when you install newer versions the settings carryover to the next but not in reverse, I guess. 
The icon for the 4.4.1 came back as a white rectangle when opened the first time & used all day then the next day it had the original light blue "C".  The 4.8 was the normal icon.

### Expected results

I don't operate the printers over the night everything is shut down, but I am on the "Insider Program" with Windows & don't know if any download that was done over night may have caused it. There have been a few downloads to my PC's the past week, but it's odd that it didn't affect the 5.2.3A version. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm sure it won't show anything on the log file because of the app being reinstalled, but haven't looked.

middlechamber, 2023-04-10:
Didn't know if anyone else has had that situation.

middlechamber, 2023-04-10:
Accident I hit the button

MariMakes, 2023-05-05:
Hey @middlechamber,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I'm not sure if I understand your problem. Are you asked to go through the Cura setup? 
[image]

Are you sure you are selecting the correct .exe for using Cura? The installation and the program have similar .exe names:
[image]

The .exe you are looking for is called UltiMaker-Cura.exe (not Cura-Engine) and should be found here: C:\Program Files\UltiMaker Cura 5.3.0-alpha

If that doesn't help, can you show some screenshots of what you are encountering?

closed 2023-05-30 by github-actions[bot]