Ultimaker/Cura #15203: Not slicing manual supports to full height

Poporacer, 2023-04-10:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Created cylindrical object with a 90-degree overhang
2. Added manual supports around the circumference.
3. Sliced the file

### Actual results

The supports are only sliced for 4 or 5 layers and then stop. I have printed a few of these that were almost identical and they sliced correctly.

### Expected results

The supports should print up to the overhang they are supposed to support.
[CE3PRO_Jack Cap v2.zip](https://github.com/Ultimaker/Cura/files/11191447/CE3PRO_Jack.Cap.v2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

File Preview
[image]

Layer 4
[image]

Layer 20
[image]

nallath, 2023-04-11:
This is caused by the "Minimum Support Area", which is set to 2mm. At the bottom, it does print them, as the tapered edge makes them go slightly above the 2mm^2 that is the default.

If you change the setting to 1mm^2, it will print the manual supports that you added.

closed 2023-04-11 by nallath

Poporacer, 2023-04-11:
Thank you, that make sense. I slightly decreased one dimension that probably put it within that minimum...Thanks a million!