Ultimaker/Cura #15207: Random Slicing Failure

bcull05, 2023-04-11:
### Application Version

5.3.0

### Platform

MacOS Monterey

### Printer

Elegoo Neptune 3

### Reproduction steps

1. Set up file with tested and functional slicing profile
2. Hit slice

### Actual results

File begins to slice, gets stuck halfway through, and then gives the message "Slicing failed with an unexpected error. Please consider reporting a bug." Sometimes just hitting slice again is successful, other times it's not.

### Expected results

File slices and is able to be previewed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Mando Helmet Top.3mf.zip](https://github.com/Ultimaker/Cura/files/11198316/Mando.Helmet.Top.3mf.zip)
[image]
[image]

MariMakes, 2023-05-05:
Hey @bcull05,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍
I tested your model in 5.3 and got the same crash. But only when I had tree supports enabled, it works with regular support. 
We are working on improving our tree supports for the upcoming 5.4 release, so this is a good test case. 

Thank you for sharing. 😊

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-25](https://ultimaker-o7.sentry.io/issues/4864528358/)