Ultimaker/Cura #15212: error ( not slicing )

APerson43, 2023-04-11:
### Application Version

5.0.0 i believe 

### Platform

windows 10

### Printer

anycubic vyper

### Reproduction steps

this error is only occurring for two specific 3d models.
when i hit slice it loads like normal then comes up with an error message saying : "slicing failed with an unexpected error. please consider reporting a bug on our issue tracker 

### Actual results

 N/A ( all in first bit )

### Expected results

N/A  ( all in first bit )

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A  ( all in first bit )

nallath, 2023-04-11:
Could you share the models that this happens with and also check if the issue is still there with Cura 5.3?

APerson43, 2023-04-11:
and how do i check my version?

nallath, 2023-04-11:
> how do i share the model with you?

If you zip the models you can add them to a message.

> and how do i check my version?

You can do that by navigating to help -> about. The version should be listed in the top underneath the "Ultimaker Cura" text.

APerson43, 2023-04-11:
[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/11201965/New.Compressed.zipped.Folder.zip)

APerson43, 2023-04-11:
i also did a fresh install of cure earlier to see if that would fix it