Ultimaker/Cura #15220: Windows 11 lagging when monitor tab is selected

easterly81, 2023-04-12:
### Application Version

5.3.0

### Platform

windows 11

### Printer

ender3 v2

### Reproduction steps

Start print with octoprint plugin via network. When the monitor tab is selected windows lags. I can hardly use the mouse. 

### Actual results

Windows lags pretty bad, I can hardly use the mouse. 

### Expected results

Should not lag the whole machine. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Processor	AMD Ryzen 5 5500                                  3.60 GHz
Installed RAM	16.0 GB (15.9 GB usable)
Device ID	CD361EF8-3782-48A7-9026-438DE40A0308
System type	64-bit operating system, x64-based processor
Radeon Pro W5700

[cura.log](https://github.com/Ultimaker/Cura/files/11213663/cura.log)

rburema, 2023-04-18:
Is this also the case when the octoprint plugin is disabled?

easterly81, 2023-04-18:
It was actually my mouse so you can close the issueGet Outlook for Android

rburema, 2023-04-18:
Thanks for the reply! Will close this issue then.

closed 2023-04-18 by rburema