Ultimaker/Cura #15223: Pause at height

FabienBoissel, 2023-04-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Je suis habitué à utilise la fonction "Pause at height" (insertion d'écrou métallique) depuis quelques années, j'ai une mise à jour de Cura et du firmware de l'imprimante.

### Actual results

La fonction "pause at height " ne fonctionne plus, l'imprimante ne s'arrête pas. Les impressions que je réalise sont alors non conforme et rebuté

### Expected results

L'imprimante doit s'arrêter au layer number programmer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-04-15:
Thanks for the report.
If it is the same printer as you used before, and the pause command worked before, then it really should work now.  Was there a recent firmware update?
Maybe @nallath has some information on this?

nallath, 2023-05-01:
This was fixed for 5.3.1

closed 2023-05-01 by nallath