Ultimaker/Cura #15232: Printer plate graphic blocks bottom view of parts

tisimst, 2023-04-13:
### Application Version

5.3.0-alpha+Xmas

### Platform

Windows 10

### Printer

Anycubic Chiron

### Reproduction steps

Loaded part. Rotated view to look from below. 

### Actual results

Printer graphic blocks view of part.

### Expected results

No graphic blocking view of parts on build plate. OR an option to remove print area graphics.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-04-15:
Thanks for the report.
All of that is within your Printer Definition file which was submitted to Cura by a contributor or by AnyCubic.  Within the definition file will be a line regarding the "Platform".  It will indicate which file in the "Meshes" folder is used for your printer.  This is from the Chiron definition file.
"author": "Patrick Glatt", 
"platform": "anycubic_chiron_platform.obj",
I'll close this as it isn't an UltiMaker Cura problem but rather something you would need to take up with AnyCubic or Mr. Glatt.  I hope you understand.

closed 2023-04-15 by GregValiant

tisimst, 2023-04-17:
> Thanks for the report. All of that is within your Printer Definition file which was submitted to Cura by a contributor or by AnyCubic. Within the definition file will be a line regarding the [...]

I have no idea who the author is, but thank you for letting me know. Is the Chiron definition file something that gets compiled into Cura or is it something I can edit after the fact? 

And in regards to "this isn't an Ultimaker Cura problem", I'm not sure that's true at all. While the *design* of the graphic is not a UM/C problem, the ability to show/hide it most certainly is. If the user has *no way* to see past it, then that is something UM/C can fix. I still say this is a bug and should be at least a feature request if there currently isn't a way to easily hide the graphic. I don't understand why the graphic isn't forced to be at least semi-transparent in the first place. Very frustrating as an end user when I need to be able to see the surfaces that need supports, but I'm unable to. When viewing from the top, a graphic is very nice looking as long as the grid is still visible, but from the bottom, it's very much in the way.

Thanks for hearing me out. And know that I'm not bringing this up lightly. I've been in this industry for years doing metal additive and being able to see the parts from the bottom during support design was an absolute necessity for the vast majority of things we printed. I'm sure I'm not the only one.

tisimst, 2023-04-17:
I'm not sure how I found it today and not any time searching before, but I found where the Definition files are located and was able to make the change I needed. Thanks for the pointer!