Ultimaker/Cura #15234: 5.3 Initial layer speed doesn't apply to supports

sangdrax8, 2023-04-13:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender-5 S1

### Reproduction steps

Slice a model with the initial layer speed low (like 10-20 mm/s) while leaving the infill speed faster.  The first layer of the supports should be the same speed as the rest of the first layer.

### Actual results

All my supports that printed after the main piece are ignoring the first layer speed and printing to fast.

### Expected results

The supports should follow the first layer speed on their first layer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I have run this exact same project file under 5.2.1 and with out any changes it slices correctly.
[dragon_knocker.zip](https://github.com/Ultimaker/Cura/files/11226913/dragon_knocker.zip)

sangdrax8, 2023-04-13:
As I am now watching it print with my 5.2.1 slice, I am seeing that it is building all the supports before the main first layer.  So it would seem this issue stems from the fact that not all my supports in 5.3.0 were being printed before the first layer.  Perhaps they are even only printed in the 2nd layer, which would be even worse than printing to fast.

GregValiant, 2023-04-14:
Thanks for the report.
I sliced your project file and the Initial Layer Print Speed looks correct.  It's actually printing at a speed below your set speed of 20mm/sec.
As you noted, the real problem is that the support doesn't appear on the first layer.  Instead I get this...
[image]

That little squiggle doesn't get duplicated on any other layers.  Here is the second layer and we see that the support has started.
[image]

There are only 103 lines of gcode for the Initial Layer Support but there are 394 lines for the support on the second layer.  The model itself seems to have sliced OK
Somebody from the Cura team will take a look.