Ultimaker/Cura #15235: スライスが完了しない

castasia, 2023-04-14:
### Application Version

5.3.0

### Platform

windows10 pro

### Printer

ultimaker S5

### Reproduction steps

添付ファイルのモデルのスライスを実行しても完了しません
積層ピッチ0.15[mm] 材料 PLA
[image]

### Actual results

積層ピッチ 0.2[mm]の場合　時々スライスは完了する
積層ピッチ 0.15[mm]の場合スライスできない

### Expected results

スライスを完了するために必要な設定等なりましたら教えてください
[shade 2002.zip](https://github.com/Ultimaker/Cura/files/11228323/shade.2002.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[shade 2002.zip](https://github.com/Ultimaker/Cura/files/11228336/shade.2002.zip)

GregValiant, 2023-04-14:
Thank you for the report.
I didn't have any trouble slicing the model.  I tried right side up and upside down.  I'm unable to duplicate the problem.
A project file would be better because it will contain your printer, the model, and your settings.  Load the model, set Cura up so slice, and the use the "File | Save Project" command, zip the 3mf file and post the zip folder here.
[image]

castasia, 2023-04-19:
テストして頂いたことに感謝します
私の方でも積層ピッチを0.4[mm]に設定した時は　スライスは問題なく完了します
0.2[mm]の時も時間は必要ですがスライスは完了します
しかし積層ピッチを0.15[mm]とした時 スライスは「予期しないエラーが発生し　スライスに失敗しました」となってしまうのです

GregValiant, 2023-04-19:
I have been changing the Initial Layer Height and the Layer Height around.  It slices at all of the combinations I have tried.