Ultimaker/Cura #15236: Slow and Sluggish Scaling menu when Win11 Text Cursor Indicator acccessibility help is enabled

xwcg, 2023-04-14:
### Application Version

5.2+

### Platform

Windows 11

### Printer

Anycubic

### Reproduction steps

1. Open Windows Settings
2. Navigate to Accessibility -> Text Cursor
3. Enable "Text cursor indicator"
4. Open Cura
5. Import an STL file
6. Switch to the scaling menu
7. Attempt to change the model scale

### Actual results

Cura practically non-responsive, changing single numbers takes minutes

### Expected results

Changing numbers should be instantaneous

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

When the cursor is disabled, it works as intended, even if disabled at runtime.

fieldOfView, 2023-04-14:
Could you try this nightly build? 
https://github.com/Ultimaker/Cura/discussions/15098

This sounds similar to a bug people that use the Sidebar GUI plugin or the Settings Guide plugin have reported in combination with the FancyZones powertoy. For me that bug is fixed with the nightly build, though sofar I have not gotten confirmation from the people that reported the issue.

xwcg, 2023-04-14:
That seems to fix it for me! I'll leave it open though? Not sure how you guys handle closing issues.