Ultimaker/Cura #15240: Permission denied to write gcode file to WebDav Network drive

heinerolli, 2023-04-14:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

Connect a webdav Server to your Windows PC
Verify that you can write / copy files to this WebDav drive via std. Windows File Explorer (works for me)
Try to save ('save to disk') a gcode file to this network drive (or its subfolders)

### Actual results

The gcode file is not saved on the webdav Networt Drive - (I am able to save the file to any other (non Webdav) Network drive)

The error I get is "Permission denied when trying to save ...."

When I start Cura in Admin mode, I don't even see the Webdav Drive within Cura!

### Expected results

The gcode file should be saved on the Webdav Networt Drive - similar as if you would save it to any other (Network)location 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

heinerolli, 2023-04-14:
Nevermind & sorry.... Can be closed. 
My stated problem can be resolved by enabling administrator rights for current user in windows....
Like i.e. here shown: https://www.youtube.com/watch?v=920Gu8uxX9I

heinerolli, 2023-04-14:
Sorry again......
This issue is NOT resolved - it just looked as if it worked when you try to store a gcode file with a filename which is already on the drive: Then it asks you if overwite is ok and it pretends to overwrite it - but it doesn't. (But in this case, no error message comes up)

The original statement is still true: When you want to store a new file on the webdav drive via cura, you get the error message from the picture and nothing is stored.

MariMakes, 2023-05-08:
Hey @heinerolli,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I brought your issue up with the team. 
What you are describing is outside my field of expertise but they asked me to ask you the following questions.

1. Are you getting the same faulty behavior in our two ways of saving? 
They should be: 1. The button in the bottom right corner and 2 by clicking File in the top left corner, and then Export.
2. Can you confirm if the Windows Account you are using has the correct writing permissions? 

Thanks 🙏

heinerolli, 2023-05-08:
Hi; thanks for getting back to me! 
Regarding your questions:
1) I double checked and tried both ways of saving and both show the error in the exact same way:
I can not save new files, but I can overwrite an existing file on the drive. (So my current workaround is naming my print files always the same.)
2) The windows account I am using has the correct permissions. (As stated, I can copy files to the WebDav drive in question i.e. from windows file manager - just not from Cura.
Hope this helps - and thanks for looking into this! Would be nice, if this could be fixed :-)

MariMakes, 2023-05-12:
Hey @heinerolli,

I showed your answers to the team but we are a bit at a loss on how to help here. 
We don't have the setup to reproduce or fix the issue on our side. 

We believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
If you want to take a look, this would be a good place to start. It's the part that's in charge of saving the gcodes. https://github.com/Ultimaker/Cura/tree/main/plugins/GCodeWriter

I'll be closing this issue, I hope you understand.

closed 2023-05-12 by MariMakes