Ultimaker/Cura #15241: Failed to slice unknown error

JMFerg, 2023-04-14:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

LGMaker Artist Pro D

### Reproduction steps

Trying to slice large files
Won't complete slice error says unknown error
Happens very often, I end up removing items from the build plate and re-slicing, usually have to end up with only one part before it will succeed.
I have 128gb ram but issue also happened when I had only 16gb ram

### Actual results

popup says unknown error caused the slicing to fail.
[image]

### Expected results

I would expect it to slice the file, having to slice one part at a time might be great for saving filament on failed prints but is bad for set it and forget it printing of large jobs with large number of parts
[CFFFP_Right_Fin.zip](https://github.com/Ultimaker/Cura/files/11234271/CFFFP_Right_Fin.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11234278/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

files attached

GregValiant, 2023-04-15:
Thanks for the report.
I can duplicate this problem.  In the log file is a memory error _"Backend exited abnormally with return code 3221225477!"_.
That error had been coming up a few times in 5.2.2 and it was worked on.  It's a problem with the Cura Engine rather than the Cura front end.
I rotated one of the fins and the project sliced.  Someone from the Cura team will take a look.
[image]

nallath, 2023-04-25:
Thanks for reporting this! We're aware of an issue that causes some models to not slice. We don't have a fix yet, but we are investigating it.

In some cases it can help to move / rotate the model a bit. Although it's not a real fix, you might try this workaround for the moment