Ultimaker/Cura #15244: The choice of the G-Code option is not saved

passkal, 2023-04-14:
### Application Version

5.3.0

### Platform

windows 10

### Printer

KingRoon KP3S-PRO

### Reproduction steps

The choice of the G-Code RepRap option is not saved after closing the settings window. If you open it again, there will be a Marlin option again

### Actual results

The G-Code RepRap option is not saved in the printer Settings window.

### Expected results

The selected option must be saved

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/51381842/232146229-b703c1f8-ca1e-4cc0-9f6c-3d814c7dce79.mp4

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/11236714/cura.log)

passkal, 2023-04-14:
Manually to save this setting, you need to register the line in the file:

C:\Users\User\AppData\Roaming\cura\5.3\definition_changes\YOU_NAME_PRINTER_settings.inst.cfg

machine_gcode_flavor = RepRap (RepRap)
[image]

GregValiant, 2023-04-14:
Thanks for the report.
This has been reported before so I'll mark it as a duplicate of #15149 .  It isn't the only problem with that dialog box.  You cannot change the shape of the bed either.
If you are feeling brave you can open your printer definition file in a text editor and scroll down to the **"overrides":** section and add the middle line shown below.
        "machine_steps_per_mm_z": { "value": 800 },
        **"machine_gcode_flavor": { "default_value": "RepRap (RepRap)" },**
        "retraction_amount": { "value": 1 },
It's a script so understand that a "tab" is not a "space" so the indentation and the commas must be correct or you will get an error.
If you make that change then you can add a "new" printer and it will be RepRap.

passkal, 2023-04-15:
> Thanks for the report. This has been reported before so I'll mark it as a duplicate of #15149 . It isn't the only problem with that dialog box. You cannot change the shape of the bed either. If you [...]

Sorry, but you could not specify the location of this file.
I reviewed all the files, but I didn't find the word "overrides" in any of the files.

passkal, 2023-04-15:
I managed to change the type of code when I entered the line:

[values]
...
machine_gcode_flavor = RepRap (RepRap)
...

to the file:
YOU_NAME_PRINTER_settings.inst.cfg

located in:
C:\Users\User\AppData\Roaming\cura\5.3\definition_changes
[image]

Vandresc, 2023-04-28:
Hi @passkal, the new Cura 5.3.1 contains the fix for this for a number of slicing crashes. 
Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.1
Please reopen the ticket if the issues still occurs for you, thank you!

closed 2023-04-28 by Vandresc