Ultimaker/Cura #15246: Can not slice, only save to disk when I import a project.

Baer90, 2023-04-15:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3 max

### Reproduction steps

1. Import a project (3MF file) 
2. changed some settings
3. button says save to disk, but I cannot save because it isn't sliced. 

Maybe the problem is that the project is Cura 5.0.0 and it doesn't recognise the slice

### Actual results

[image]

### Expected results

Being able to slice the model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Lantern main2l.zip](https://github.com/Ultimaker/Cura/files/11238762/Lantern.main2l.zip)

Baer90, 2023-04-15:
I'm experiencing a lot of bug I have not experienced before
The skirt is no longer a nice outline, it is a parallelogram with a diagonal line through the center. Solution, remove skirt. 
In the bottom layer is a hole, diameter 70mm. In the print it has no walls, just a blob of pla in the top. Possibly my fault for enabling experimental features but the skirt thing is infureating.

rburema, 2023-04-18:
Can you please attach the log files? It looks like Cura 'thinks' it has sliced something.

Concerning the skirt problem: In general, please open new issues for separate ones. (That said, does this also reproduce with the given project file?)

MariMakes, 2023-05-05:
Hey @Baer90,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
That's a pretty cool model 🤩

I followed your exact instructions but I was not able to reproduce your issue. 
I think what you are seeing might be related to the material you selected, PLA+ ESun. 
That materials doesn't have any good profiles. 

This is what the print job looks like if you slice for PLA. 
[image]

Is it an option for you to slice with PLA instead of PLA+? 
Do you still have an issue if you slice with a different materials?

If that's not the case, feel free to let us know and we'll take another look. 
Please make sure you include very precise Steps to Reproduce.
Thanks 🙏

closed 2023-05-30 by github-actions[bot]