Ultimaker/Cura #15248: Slicing fails

FlashMoonUK, 2023-04-15:
### Application Version

5.3.0

### Platform

MacOS 13.3.1

### Printer

Dremel 3D45 with plugin

### Reproduction steps

Loaded generic profile for Dremel ECO - ABS
Loaded Model
Reorientated by 90 degrees
Support = Normal
Sliced Model - Slicing Failed
cleared build plate
reloaded model
Sliced model - OK
cleared build plate
reloaded model
Reorientated model by -90 degrees
Sliced model - failed
Switched off support
Sliced model - failed
Switched on support - Tree
Sliced model - failed
[D_Garmin to Wahoo adapter v12.3mf.zip](https://github.com/Ultimaker/Cura/files/11239233/D_Garmin.to.Wahoo.adapter.v12.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11239256/cura.log)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/112392
[cura.log.zip](https://github.com/Ultimaker/Cura/files/11239260/cura.log.zip)
59/cura.log.zip)

### Actual results

Slicing Failed with unexpected error

### Expected results

Should have sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Model Sliced yesterday with a different material profile
Have not made any changes to the model 
Have gone back to original material profile and other profiless and still get the slicing fault

GregValiant, 2023-04-15:
Thanks for the report and the files.
I am unable to get it to "not slice" evernthing I tried (moving, rotating, changing the Initial Layer Height) didn't have any effect...it still sliced.
Someone from the Cura team will look at it.

FlashMoonUK, 2023-04-15:
Thanks Greg

I have rebooted the Mac and it still fails to slice in the vertical orientation - maybe I should reload Cura

Have a good weekend

Chris

nallath, 2023-04-25:
Thanks for reporting this! We're aware of an issue that causes some models to not slice. We don't have a fix yet, but we are investigating it.

In some cases it can help to move / rotate the model a bit. Although it's not a real fix, you might try this workaround for the moment

FlashMoonUK, 2023-04-25:
Hi JaimeThe bug is caused because the minimum speed is set to 0 mm/s. This is due to the Dremel plugin. Setting it to 1.0 mm/s overcomes this problem Kind regards ChrisSay more with less.