Ultimaker/Cura #15249: Buildplate adhesion is not generated properly

andreikovari, 2023-04-15:
### Application Version

5.3.0

### Platform

Win10

### Printer

Anycubic Kossel Linear Plus

### Reproduction steps

When slicing a file, when selecting brim or skirt, the contours for both has a missing gap. 
The file was sliced in a previous version of cura and worked fine.
The gap in the buidplate adhesion seems to be dependent on the location of the piece in reference to the origin.
1. Add file to slicer.
2. Slice with buildplate adhesion enabled. 

### Actual results

Brim line consistency dependent on position of object on the buidplate
[image]
[image]
[image]
[image]
[image]

Same behavior with skirt
[image]
[image]

### Expected results

Continous line for brim or skirt

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here is the project file.
[AKLP_Body2.zip](https://github.com/Ultimaker/Cura/files/11239579/AKLP_Body2.zip)

GregValiant, 2023-04-15:
Thanks for the report and the Project File.
I can duplicate the problem with your printer and that the problem occurs with both "Brim" and "Skirt".  Changing the Bed Adhesion settings didn't help.
When I switch to my Ender 3 Pro the problem went away and I could not duplicate it with my printer.  The only real difference I see is the shape of the bed.
Someone from the Cura team will take a look.

andreikovari, 2023-04-15:
Awesome! Thanks! Let me know if I can assist you with anything else.

PerlBoy79, 2023-04-30:
Hi, I am having the same issue. I simply created a 500 brim line count (similarly skirt with 300 lines), and the missing shape is consistently there. So far happens to any objects placed. 

Current workaround is to move object away from that location or place big enough an object's 1st layer to cover that zone.

Other posts:
[https://github.com/Ultimaker/Cura/issues/14873](url)
[https://github.com/Ultimaker/Cura/issues/14876](url)
Also Anycubic Kossel Linear Plus models

**Application Version**
5.3.0 / 5.3.1 
(issues present in both versions)

**Platform**
Win11

**Printer**
Anycubic Kossel Linear Plus

[image]

yzheka, 2023-05-01:
same for me for kossel. looks like issue is only for delta printer. i have no issue for voron 0.1

fieldOfView, 2023-09-01:
Probably related to the 'Origin at center" setting and how CurEngine recently started clipping the skirt and brim to the buildplate extents.

GregValiant, 2023-09-01:
This was addressed with [PR #15847](https://github.com/Ultimaker/Cura/pull/15847/commits/8cfa59eabf7c140e99e9babe26eb5e8a80038a05) .  It has been merged and the problem should be fixed in the next release.

Yes, it did look like the brim/skirt "clipping" was the issue but it was the way the "disallowed areas" were defined in the printer definition files that was causing the problem.  

I'll go ahead and close this as it has been addressed.

closed 2023-09-01 by GregValiant