Ultimaker/Cura #15250: Slicing Failed: Please Report Bug

bricomp, 2023-04-15:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Elegoo Neptune 3 Plus

### Reproduction steps

Attempted to slice part.

### Actual results

Cura reported Slicing failed with an unexpected error.
[image]

### Expected results

Expected a sliced gcode file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[EN3P_Thermostat Enclosure Lid.zip](https://github.com/Ultimaker/Cura/files/11239599/EN3P_Thermostat.Enclosure.Lid.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11239600/cura.log)

GregValiant, 2023-04-15:
Thanks for the report.
It wouldn't slice for me in the orientation you had it in.  I rotated it 90° and it sliced.
I'll mark this as a slicing error and a duplicate as this behavior has been noted before.  It was worked on for 5.3 but it doesn't look like it's fixed yet.
Someone from the Cura team will take a look.

In the meantime, you need to reset your profile between models.  Your other bug report showed you had Spiralize turned on and so does this one.  In addition, if you were to print this with the topside down it wouldn't need supports.  If you want to print it right side up then I think you should use regular supports.  Turning Spiralize off also turns off "Remove All Holes".
Making those two changes (Spiralize off and Regular Supports enabled) this model sliced just like you have it oriented.

bricomp, 2023-04-16:
“if you were to print this with the topside down it wouldn't need supports. If you want to print it right”

I wanted to use the Ironing Feature on the top.