Ultimaker/Cura #15251: Slicing fails after blocking supports

bernardhny, 2023-04-15:
### Application Version

5.3.0

### Platform

Windows 11 Pro

### Printer

Creality CR10 S Pro V2

### Reproduction steps

Placing support blockers on 3 parts of the model. 
Selected Slice
Slicer failed after about 5 seconds
Repeatable.

### Actual results

[image]

### Expected results

Slicer finished slicing so I could continue to place and block supports as necessary

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Attached is project file. I am trying to block areas of support, it fails to slice. How do y ou not support project files from Cura? Makes no sense.

GregValiant, 2023-04-15:
Thanks for the report.
The website is hosted by GitHub and it is their rule that only obvious TXT files can be uploaded as is.  Everything else needs to be zipped.  It probably has something to do with storage space on their servers.
Without a project file and a Cura.log file there can't be any investigation.  If you need this looked at you will need to provide those files so we can tell what's going on.

bernardhny, 2023-04-15:
[CCR10SPRO_Six_fingered_God_of_secrets_with_pedestal.zip](https://github.com/Ultimaker/Cura/files/11240755/CCR10SPRO_Six_fingered_God_of_secrets_with_pedestal.zip)