Ultimaker/Cura #15254: Cannot run Cura from Source / Ubuntu 22.04.2 LTS x86_64 / fails to conan build.

erickCantu, 2023-04-15:
### Application Version

5.3

### Platform

Ubuntu 22.04.2 LTS x86_64

### Printer

Sovol 06

### Reproduction steps

I tried to use both AppImages  for Linux without success in Ubuntu 22.04.  When selecting the materials menu fro the configuration, Cura brakes. The error report was sent through Cura. 
Then I looked to run it from source. I got mistakes related to missing libraries. However, these libraries are installed. There are a couple of warnings, but I believe these are not related to the problem.  Any advice to make it run? Thanks in advance.

- Follow the steps described at: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source including:

1. Ensure the requirements are meet for Linux and update as required. 
2. Source .venv with Python 3.10.6
3. Make Cura directory, and  Clone Cura GitHub repository in it.
4. Run in zsh: conan install . --build=missing --update -o cura:devtools=True
5. The installation brakes

### Actual results

### Check Requirements
```bash
╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ gcc-12 --version
gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ python --version
Python 3.10.6

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ pip install sip 
Requirement already satisfied: sip in ./.venv/lib/python3.10/site-packages (6.7.8)
Requirement already satisfied: packaging in ./.venv/lib/python3.10/site-packages (from sip) (23.1)
Requirement already satisfied: ply in ./.venv/lib/python3.10/site-packages (from sip) (3.11)
Requirement already satisfied: setuptools in ./.venv/lib/python3.10/site-packages (from sip) (63.2.0)
Requirement already satisfied: tomli in ./.venv/lib/python3.10/site-packages (from sip) (2.0.1)

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ cmake --version
cmake version 3.26.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ ninja --version
1.10.1

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
╰─○ autoreconf --version
autoreconf (GNU Autoconf) 2.71
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<https://gnu.org/licenses/gpl.html>, <https://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.

╭─erick at erick-MS-7C35 in ~/Applications/Cura using «.venv»
[... 410 lines cut ...]
### Check that libx11-dev exists
``` bash
╭─erick at erick-MS-7C35 in ~/Applications/Cura/Cura on main✔ using «.venv»
╰─± sudo apt-get install libx11-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libx11-dev is already the newest version (2:1.7.5-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
╭─erick at erick-MS-7C35 in ~/Applications/Cura/Cura on main✔ using «.venv»
╰─± 
```

### Expected results

A full installation of Cura from Source.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional information.

rburema, 2023-04-18:
I'm on Windows at the moment, but reading the error it seems as if 'The following packages need to be installed' refers to _all_ packages needed for that step. As such, I think the ones actually missing are just above that list, in other words, all of the `dpkg-query: no packages found matching <package>` messages. Can you check if any of those are installed?

MariMakes, 2023-05-05:
Hey @erickCantu,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

I'm super curious. What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

degid, 2023-05-07:
I had the exact same problem on Arch Linux. Unfortunately, the message does not indicate the missing package. I just simply fed this whole list to pacman :)

erickCantu, 2025-02-27:
I am sorry to come back this late.  The problem was a bad RAM stick. It did not showed errors in Windows and everything else in Linux was working well, until id didn't. After changing the RAM, the problem was solved. 
So check your RAM, specially if is overclocked.