Ultimaker/Cura #15260: Rotate in Objects original orientation

lexx9999, 2023-04-16:
### Is your feature request related to a problem?

So I have objects that are too big for placing them along the usual axes. So I place them diagonal on the plate.
Problems on this:
- Automatic placement doesn't try diagonal placement (I'm ok with that)
- After rotating it if the object placement for the support is not good, I can only rotate along the standard axes

Result is a tilted object, or I have to reload the object an restart from scratch.

Provided a simplified example (yes, I know this object could be flipped, but it's for illustration purposes).

### Describe the solution you'd like

I would like to (Preference in descending order)
1. Toggle between Rotation of the Objects-Original Axes and the Printers-Area-Axes.
Natural behavior would be either to press "space" in rotation mode or having another symbol in the rotate "menu" next to "reset" "lay flat" 
2. "lay flat" not rotating the object but keeping the orientation

### Describe alternatives you've considered

Reloading file, adjusting angle to horizontal plane, then rotate about 45° to fit to my needs.

### Affected users and/or printers

Everyone who wants to print objects on printers that fit diagonal but not along base axes. 
Especially when there should be "nice surfaces" on a specific side.

### Additional information & file uploads

Diagonal Alignment:
[image]
"Future Object's Axes based rotation" (Gray: Current rotation)
[image]

Vandresc, 2023-04-20:
Welcome to the UltiMaker Cura Github 🚀

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

Vandresc, 2023-04-28:
Hi @lexx9999, we are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we believe we will not work on this anytime soon and will therefore defer this issue, we think it is quite tricky to make this behavior consistent, at least for now. 
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2024-06-08 by github-actions[bot]