Ultimaker/Cura #15263: Cura hangs completely during "add printer" wizard

faern, 2023-04-16:
### Application Version

5.3.0 (and 5.4.0-alpha)

### Platform

Fedora Linux 37

### Printer

Voron 2.4 350mm (and many others)

### Reproduction steps

1. Clean install of Cura 5.3.0 from either flatpak or the AppImage
1. Go to "Add printer" -> "Non UltiMaker printer" -> "Add a non-networked printer"
1. Select almost any printer (I have reproduced this hang with "Ultimaker 2", "Voron 2 350mm" and "Prusa i3") and hit next.

### Actual results

Cura will lock up completely at this stage. Or if lucky it will add the printer, go back to show the print bed and then lock up a few seconds later.

### Expected results

I should be able to add printers without issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm on Fedora 37. I have tried:
* Both the AppImage file from github as well as installing and running from flatpak. Exact same results
* Both in Wayland and X11 window mode, no difference. (Yes, I tried with `QT_QPA_PLATFORM=xcb`.
* The Cura 5.4 alpha build in https://github.com/Ultimaker/Cura/discussions/15098. Still same results.

There is no `cura.log` stored at `~/.local/share/cura/5.3/cura.log`. I don't know what happened. I had one, but then I deleted it to get a new fresh log to upload. But now no new logs are generated.

When running from a terminal I have the following output from Cura. But all this output is printed long before the lockup:
```
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaL9H4Qv/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaL9H4Qv/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///tmp/.mount_UltiMaL9H4Qv/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaL9H4Qv/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
```

faern, 2023-04-17:
I tried the flatpak install on two other machines now. One with Fedora 36 and one with fedora 37. It works fine on both. So not sure what's broken on my main desktop machine. Could it be something regarding GPU drivers? Because the desktop has a dedicated AMD graphics card, but the other two machines just have intel graphics.

nouspiro, 2023-04-18:
I think I have same issue. It is quite possible that it get stuck in USB printing. Because after deleting config I managed to get it working once. But after one reboot I get this message and right after that it get stuck. That one time that it worked I could see that it was trying different baud rates on /dev/ttyS0
`
[JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started`

After I run `sudo chmod 600 /dev/ttyS0` cura started correctly.

eloo, 2023-04-29:
Same for me. 
The wizard is freezing when i try to add my printer.
Here are the logs

```
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///opt/ultimaker-cura/share/cura/resources/qml/Actions.qml" and "file:///opt/ultimaker-cura/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///opt/ultimaker-cura/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///opt/ultimaker-cura/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"

** (UltiMaker-Cura:39963): WARNING **: 21:06:33.020: atk-bridge: get_device_events_reply: unknown signature
QObject::startTimer: Timers cannot be started from another thread
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/SimpleCheckBox.qml:24:5: QML SimpleCheckBox: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML PrintHeadMinMaxTextField: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/ComboBoxWithOptions.qml:24:5: QML ComboBoxWithOptions: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
file:///opt/ultimaker-cura/share/cura/resources/qml/MachineSettings/NumericTextFieldWithUnit.qml:21:5: QML NumericTextFieldWithUnit: Binding loop detected for property "width"
```

nouspiro, 2023-04-30:
Try run `sudo chmod 600 /dev/ttyS0` this workaround helped me.

eloo, 2023-04-30:
I have tried this already but did not work.. also when I run cura as sudo it's not working. 

Further my printer is not connected to my PC

MariMakes, 2023-05-05:
Hey @faern,

Thanks for your super detailed report 👍
I'm on Windows myself but we have some Linux Developers that might have a better grip on what is happening. 
I'll bring it up to the team to see what they can do. 
Fingers crossed 🤞

eloo, 2023-05-19:
@MariMakes 
any updates here since the last two weeks?

It really a shame that the Linux support is that bad for Cura.
With nearly every minor update of Cura its not working again :(

faern, 2023-05-22:
> Try run `sudo chmod 600 /dev/ttyS0` this workaround helped me.

This does not work for me. But I'm not ruling out the possibility that it's a USB issue in some way. One of the differences between my working and my non-working computer is the hardware/connected peripherals.

piontec, 2023-06-22:
Same here. `5.2.1` was the last release that worked for me. `5.3.x` and `5.4.0-beta` have the same behavior: when I try to add my ender 3 (without the printer being connected to the PC, as it's a wireless with octoprint), the "add printer" dialog freezes and cure is dead. Tested with adding it as 'ender 3/ v2' and 'ender 3 pro'. I'm on ubuntu 23.04 with wayland.

MariMakes, 2023-06-26:
Hey @piontec,

I recognize your Ubuntu version 🤔
Ubuntu does not ship FUSE 2 anymore on 23.04. This is required to run our appimages.
We have reports that Cura 5.4 runs if you install FUSE 2 with `sudo apt install libfuse2`

Can you let us know if installing this library works for you?

eloo, 2023-06-27:
Still not working with 5.4.0-beta.1-linux-modern

system: manjaro
linux: 6.1

logs from fresh (no ~/.config/cura and ~/.local/share/cura dirs)
```
                                                                                                                                                            14:32:49   
trimesh/curvature.py:12: DeprecationWarning: Please use `coo_matrix` from the `scipy.sparse` namespace, the `scipy.sparse.coo` namespace is deprecated.
  from scipy.sparse.coo import coo_matrix
UM/Qt/QtApplication.py:697: DeprecationWarning: QEvent constructor is deprecated
  super().__init__(self.QtFunctionEvent)
2023-06-27 14:31:50,804 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
2023-06-27 14:31:50,804 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.4.0-beta.1
2023-06-27 14:31:50,804 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.4.0
2023-06-27 14:31:50,805 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None
2023-06-27 14:31:50,805 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/uranium must be a folder.
2023-06-27 14:31:50,805 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/uranium must be a folder.
2023-06-27 14:31:50,805 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/uranium must be a folder.
2023-06-27 14:31:50,805 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/.mount_UltiMaoYKRN8/plugins must be a folder.
2023-06-27 14:31:50,805 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/uranium/plugins must be a folder.
2023-06-27 14:31:50,806 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/Resources/cura/plugins must be a folder.
2023-06-27 14:31:50,806 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
2023-06-27 14:31:50,806 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /home/joseph/.config/cura/5.4
2023-06-27 14:31:50,806 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /home/joseph/.local/share/cura/5.4
2023-06-27 14:31:50,806 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /home/joseph/.cache/cura/5.4
2023-06-27 14:31:50,807 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [536]: Found config: ['/home/joseph/.config/cura'] and data: ['/home/joseph/.local/share/cura']
2023-06-27 14:31:50,807 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [540]: Latest config path: None and latest data path: None
2023-06-27 14:31:50,807 - INFO - [MainThread] UM.Qt.QtApplication.initialize [164]: Preferences file not found, ignore and use default language 'en_US'
2023-06-27 14:31:50,807 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaoYKRN8/share/uranium/resources/bundled_packages/uranium.json
2023-06-27 14:31:50,807 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaoYKRN8/share/cura/resources/bundled_packages/cura.json
2023-06-27 14:31:50,807 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
2023-06-27 14:31:50,808 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile
2023-06-27 14:31:50,808 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile
2023-06-27 14:31:50,808 - INFO - [MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...
2023-06-27 14:31:50,808 - INFO - [MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...
2023-06-27 14:31:50,808 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltiMaoYKRN8/share/uranium/resources/bundled_packages/uranium.json
2023-06-27 14:31:50,809 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /tmp/.mount_UltiMaoYKRN8/share/cura/resources/bundled_packages/cura.json
2023-06-27 14:31:50,809 - INFO - [MainThread] UM.PackageManager._loadManagementData [203]: User package management file /home/joseph/.local/share/cura/5.4/packages.json doesn't exist, do nothing
2023-06-27 14:31:50,809 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 
2023-06-27 14:31:50,809 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib/cura must be a folder.
2023-06-27 14:31:50,809 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib64/cura must be a folder.
2023-06-27 14:31:50,809 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/libx32/cura must be a folder.
2023-06-27 14:31:50,810 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /tmp/lib32/cura must be a folder.
2023-06-27 14:31:50,810 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin ConsoleLogger 1.0.1
[... cut ...]
2023-06-27 14:32:18,781 - DEBUG - [MainThread] cura.Machines.Models.IntentSelectionModel._update [73]: Updating IntentSelectionModel.
2023-06-27 14:32:18,957 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2023-06-27 14:32:19,045 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [164]: New error check scheduled.
2023-06-27 14:32:19,141 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [221]: Error check finished, result = False, time = 0.10s
```

eloo, 2023-07-09:
@MariMakes any updates here?

nallath, 2023-07-14:
My best guess is that it is related to the USB printing plugin, since it only seems to trigger for machines that have the "supports_usb_connection" as far as I can tell.

To validate my suspicion, can people check if adding an Ultimaker 3, s3 or s5 also cause the issue? Since I don't have the issue myself, i can't check it myself :(

nallath, 2023-07-14:
> @nallath you're the best!
> 
> it really looks like this is the issue!
> 
> i have added
> 
> ```
>         "supports_usb_connection": false
> ```
> 
> to my printers profile `creality_ender3pro.def.json` and now its working
> 
> so please guys... fix the usb printing plugin or at least give an option to disable the broken plugin!

Well, you can disable it in the marketplace. The problem is that I can't reproduce it (but I'm on Ubuntu, so maybe that's it?), which makes it very hard to figure out what is going on and how to fix it. 

I've created a PR with some extra logging, i hope that will provide the info we need to fix it.

jellespijker, 2023-07-14:
The PR with the extra debugging statements are merged to main and I started a build for Linux such that you guys can run with this version. Sharing these logs would help @nallath out with figuring out what goes wrong.

You can download these artifacts here: https://github.com/Ultimaker/Cura/actions/runs/5552874115

eloo, 2023-07-14:
@jellespijker @nallath 
here are some looks for usb

```
/tmp/.mount_UltiMapR0cxP/share/cura/plugins/USBPrinting/USBPrinterOutputDevice.py:280: DeprecationWarning: invalid escape sequence '\d'
2023-07-14 13:58:47,502 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin USBPrinting 1.0.2
2023-07-14 13:58:52,825 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.addCallback [31]: on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x7f9326430700>>] added.
2023-07-14 13:58:52,827 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.addCallback [31]: on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x7f932674beb0>>] added.
2023-07-14 13:58:52,828 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started for /dev/ttyS5
2023-07-14 13:58:52,831 - DEBUG - [JobQueueWorker [2]] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started for /dev/ttyS4
```

eloo, 2023-07-14:
but so far

**workaround**: Disable "USB Printing" in marketplace

nallath, 2023-07-14:
Can you run 
`sudo dmesg | grep ttyS`

And share the output?

eloo, 2023-07-14:
```
[    0.441936] 0000:02:00.1: ttyS4 at I/O 0x3200 (irq = 33, base_baud = 115200) is a 16550A
[    0.442202] 0000:02:00.2: ttyS5 at I/O 0x3100 (irq = 34, base_baud = 115200) is a 16550A
```

FYI: i don't have my printer connected using USB... i use octoprint, so the printer was and will never be connected using usb

nallath, 2023-07-14:
Yeah, i figured. But I'm trying to figure out why the auto detect is hanging itself up on some machines and not others.

eloo, 2023-07-14:
@nallath maybe it would be enough to set a timeout for the autodetect to just a few seconds

piontec, 2023-07-14:
I can confirm that adding ultimaker 3 worked without issues, only adding my networked ender 3 with octorpint seems to be an issue

nallath, 2023-07-17:
> @nallath maybe it would be enough to set a timeout for the autodetect to just a few seconds

It already has a timeout. The auto-detect also happens in a separate thread, so it shouldn't block the application itself. I'm at a bit of a loss what is exactly causing it to hang...

nallath, 2023-07-25:
So, one more thing that people might be able to try. It does require changing a part of the code. 

Can people check if removing these lines

```python3
        try:
            programmer.connect(self._serial_port)
            serial = programmer.leaveISP()
        except ispBase.IspError:
            programmer.close()
```

make the problem go away.

MarcoCLA, 2023-08-16:
Yes it does work when commented.
[cura.broken.log](https://github.com/Ultimaker/Cura/files/12357232/cura.broken.log)
[cura.commented.log](https://github.com/Ultimaker/Cura/files/12357233/cura.commented.log)

wawanbreton, 2023-08-29:
Hi @MarcoCLA,
I tried to confirm your statement, but couldn't reproduce the bug either :(
My guess is that something on your system is messing with Cura trying to handle the usb port, either a service (like ModemManager) or a USB device (either external or internal) which would partially respond to our auto-detection calls.

Would you mind sharing the output of the following commands :
`lsusb`
`lsmod`
`ps aux`
`ls  -lh /sys/class/tty/*`

For the record, we have added a ticket to the backlog with the intent to improve this.
For internal reference - CURA-10972
Thanks for the report! :+1:

MarcoCLA, 2023-09-01:
Two devices cause it to hang on my system.

`marco@MC-ThinkPad ~ % ls -lh /sys/class/tty/* | grep -E 'ttyS5|ttyS4'`
`lrwxrwxrwx 1 root root 0 Sep  1 07:54 /sys/class/tty/ttyS4 -> ../../devices/pci0000:00/0000:00:01.4/0000:03:00.1/tty/ttyS4/`
`lrwxrwxrwx 1 root root 0 Sep  1 07:54 /sys/class/tty/ttyS5 -> ../../devices/pci0000:00/0000:00:01.4/0000:03:00.2/tty/ttyS5/`

`marco@MC-ThinkPad ~ % lspci | grep -E '03:00.1|03:00.2'`
`03:00.1 Serial controller: Realtek Semiconductor Co., Ltd. RTL8111xP UART #1 (rev 0e)`
`03:00.2 Serial controller: Realtek Semiconductor Co., Ltd. RTL8111xP UART #2 (rev 0e)`

Looks like some IPMI interface.

wawanbreton, 2023-09-05:
Hi @MarcoCLA,

Thanks for the report ! You do have some serial devices on your system, which may be linked to your problem, depending on how they react to what Cura is doing at startup.

I still have to find the best way to fix this without breaking the USB printing support. In the meantime, can you try starting cura with the following environment variable :

`CURA_DEVICENAMES="$^" ./UltiMaker-Cura-5.4.0-linux.AppImage`

That should completely stop opening the USB ports, and hopefully allow you to use Cura normally.

wawanbreton, 2023-09-05:
Well, from what I can tell, @MarcoCLA your theory really seems to make sense. The method `Serial.flush` calls `tcdrain`which can indeed block forever, and as it is a system call, it can block the whole Python interpreter.

Now there are many possibles ways for us to fix this. I'll discuss them with the team to find the most appropriate one.

wawanbreton, 2023-09-11:
Hey @MarcoCLA, could you please test [this AppImage](https://github.com/Ultimaker/Cura/suites/16028704425/artifacts/914827232) ? It should fix your issue without breaking USB support.

If you can't start it with your distribution, just tell me and I'll make a new one with different options.

wawanbreton, 2023-09-12:
Sorry, the link won't work for you.. You can download the file [here](https://drive.google.com/file/d/1wf6BIdJVhF5HxAVituFCoPrmSh0JXKo2/view?usp=sharing)

MarcoCLA, 2023-09-13:
Do you also happen to have a patch file for me? Then I can apply it locally.

wawanbreton, 2023-09-14:
Oh sure, you may want to apply it manually, it is just somes lines of code removed : https://github.com/Ultimaker/Cura/pull/16709/commits/8f4dec2c9dd387317bbda48f6a6ef6fff71e04f1

MarcoCLA, 2023-10-07:
This patch solves this issue.

wawanbreton, 2023-10-09:
Great, thanks for the feedback ! It will be included in the next Cura release.

closed 2023-10-09 by wawanbreton