Ultimaker/Cura #15264: Sharpest Corner Z Seam Alignment is not placing seams at corners

Toxiguana, 2023-04-16:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

When setting Z Seam Alignment to "Sharpest Corner", Cura 5.3.0 is placing seams randomly over the surface of my print instead of at the sharpest corner as expected.  Changing the option for Seam Corner Preference does not have any appreciable impact to the behavior, with the exception of "None".  Setting Seam Corner Preference to "None" causes the seams to be aligned along the back of the print in a fashion which appears identical to "User Specified" Z Seam Alignment.

Cura version 5.2.2, produces the desired results for Sharpest Corner Z Seam Alignment when Seam Corner Preference is set to "Hide Seam", "Hide or Expose Seam", or "Smart Hiding".

### Actual results

Here is a screenshot with Sharpest Corner and Smart Hiding selected.  The Seam is placed randomly over the curved surface instead of on the corners between the curves.
[image]

Here is a screenshot with Sharpest Corner and None selected for Seam Corner Preference.  The seam is placed in a straight line on the back of the part.
[image]

### Expected results

Here is a screenshot from Cura 5.2.2 with Sharpest Corner and Smart Hiding selected.  This is the desired result where the seam is placed on corners, leaving the long arcs undisturbed.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura_5.3.0_bugreport_FatYoshi.zip](https://github.com/Ultimaker/Cura/files/11243795/Cura_5.3.0_bugreport_FatYoshi.zip)

maniaxred, 2023-04-27:
The same thing happens to me, it also happens in cure 5.3.1 recently installed, hopefully they can solve it.

MariMakes, 2023-05-05:
Hey @Toxiguana,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://ultimaker.atlassian.net/browse/CURA-10219
We also had some internal report on the topic and we have a ticket on our backlog with the intent to improve this behavior

I'll be closing this issue as a duplicate but you can follow the progress here: https://ultimaker.atlassian.net/browse/CURA-10219
Thanks for the extensive report, these make it easier for us to troubleshoot the issue ❤️