Ultimaker/Cura #15266: Wire printing fails on M1 MacBook Air

szafranski, 2023-04-16:
### Application Version

5.3

### Platform

MacOS

### Printer

Creality CR-10S

### Reproduction steps

Wire printing fails on M1 MacBook Air 

### Actual results

[image]

### Expected results

Sliced gcode

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

szafranski, 2023-05-05:
Thanks for the answer. To be honest on Windows PC it works fine. I really like this feature, especially since it is in Experimental tab, I would not remove it. I use it for some small, "fishnet"-like "sculptures. It gives some "handmade" vibe to candle holders, etc. I would even consider adding more parameters to this feature, like height of each "layer", supports etc. Regards

MariMakes, 2023-05-09:
Hey @szafranski,

Thanks for the update 😊 
I'll make sure your request ends up on the correct desk.

We believe we will not work on this anytime soon and will therefore close this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-05-09 by MariMakes