Ultimaker/Cura #15269: For many years, Cura was considered the best slicer... It was...

00raq00, 2023-04-17:
For many years, Cura was considered the best slicer... It was...

_Originally posted by @00raq00 in https://github.com/Ultimaker/Cura/issues/14958#issuecomment-1511350456_

fieldOfView, 2023-04-17:
You are entitles to your opinion, but since this post contributes nothing to Cura - or even 3d printing in general - I am going to go ahead and close this issue.

If you have an actual problem with Cura that is actionable, then please create an issue, fill in the issue template and participate in the discussion that may get the issue fixed.

closed 2023-04-17 by fieldOfView

00raq00, 2023-04-18:
> You are entitles to your opinion, but since this post contributes nothing to Cura - or even 3d printing in general - I am going to go ahead and close this issue.
>
> If you have an actual problem with [...]

or you will close issue as previous one 
https://github.com/Ultimaker/Cura/issues/14958#issue-1630997205