Ultimaker/Cura #15271: Additional Printer -- CR5 Pro-H profile needed

Blarmstrong, 2023-04-17:
### Is your feature request related to a problem?

I just got my new printer and there are not many drivers/profiles out there for my Creality CR5 Pro-H

### Describe the solution you'd like

Please add the Printer to your list of Creality Printers.

### Describe alternatives you've considered

I have tried using other printers and just renaming them but they do not have the right settings and it seems to be starving for filament. 

### Affected users and/or printers

All Creality CR5 Pro-H owners, we are all looking for this information on the various forums.

GregValiant, 2023-04-17:
The Definition files, nozzles, extruders, etc. come from a member of the community or from the manufacturer.  Creality (or their representatives) have been pretty good about submitting support files for their printers so they can be included in Cura.
If the Creality fork of Cura has definition files you could try to get them to work with Ultimaker Cura.

Blarmstrong, 2023-04-17:
I have reached out to the manufacturer for this information and have received nothing back yet. I was hopeful that maybe you already had this information or could tell me that it was a work in progress for the next software update.

Thanks,
Bill Armstrong, EMT
Plant Support Specialist
CCL Label St. Louis
636-248-8667
***@***.***
Sent from my Verizon, Samsung Galaxy smartphone

CAUTION: This email originated from outside of CCL Industries or any of its subsidiaries. Do not click links or open attachments unless you recognize the sender and know the content is safe.

GregValiant, 2023-04-18:
I looked at the Pull Requests and nothing has been submitted for that printer.  You could try using another Creality machine that is similar.  The definition files kind of automate the setup but all the settings are within Cura.  Get the bed size and firmware flavor correct in the Machine Settings dialog and you should be able to get started.

iScrat, 2024-11-13:
Did you ever get a reply...

Blarmstrong, 2024-11-13:
No I did not, there is almost no support for this printer. Oddly enough I found that the Prusa Slicer software supported this printer and I have been using this software to run the printer. Creality still sells this printer but there is almost no support out there for it, and good luck finding parts for it.

Thanks,

Bill Armstrong, EMT
Process Engineer

[CCL Label]<http://www.ccllabel.com/>
CCL Healthcare
208 Spring Drive, St. Charles, MO 63303,  USA
Cell +1 (636) 248-8667  Tel +1 (636) 573-9242  Fax +1 (636) 573-9284
***@***.******@***.***>  www.cclind.com<http://www.cclind.com>