Ultimaker/Cura #15272: Slicing randomly fails on complex model

SunnyJames, 2023-04-17:
### Application Version

5.3.0

### Platform

MacOS 

### Printer

Ultimaker 3

### Reproduction steps

Attempting slicing and fails randomly for model. 

### Actual results

Slicing fails with an unexpected error or hangs indefinitely. 

### Expected results

Model should slice. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[files.zip](https://github.com/Ultimaker/Cura/files/11253958/files.zip)
kl

GregValiant, 2023-04-17:
Thanks for the report and the files.
I can't get it to slice either.  The model is made up of three pieces so I used Mesh Tools to split it up thinking that maybe one of the sub-parts was causing Cura to quit.  Nope.
I'll mark this as a duplicate and a slicing error.  Someone from the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-26](https://ultimaker-o7.sentry.io/issues/4864594618/)