Ultimaker/Cura #15274: Slicing failed

mikerj1, 2023-04-17:
### Application Version

5.3.0

### Platform

MacOS 13.3.1

### Printer

Tronxy XY-2 Pro/Pro 2E/XY3 SE

### Reproduction steps

[terrain-163139.zip](https://github.com/Ultimaker/Cura/files/11256162/terrain-163139.zip)
Scaled to 35% (snapped), 20% lightning infill, slice
[TXY2PROPRO2EXY3SE_rawmodel-163139.3mf.zip](https://github.com/Ultimaker/Cura/files/11256214/TXY2PROPRO2EXY3SE_rawmodel-163139.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/11256218/cura.log.zip)

### Actual results

Stack trace

### Expected results

Successful slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Process:               CuraEngine [65142]
Path:                  /Applications/UltiMaker Cura.app/Contents/Frameworks/CuraEngine
Identifier:            CuraEngine
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        UltiMaker-Cura [64026]
Responsible:           UltiMaker-Cura [64026]
User ID:               501

Date/Time:             2023-04-17 18:11:57.3756 -0400
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Anonymous UUID:        D25638B4-DC05-2B48-EA3C-A80AB384D526

Time Awake Since Boot: 29000 seconds

System Integrity Protection: enabled

Crashed Thread:        10

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000038
Exception Codes:       0x0000000000000001, 0x0000000000000038

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [65142]

VM Region Info: 0x38 is not in any region.  Bytes before following region: 140723023691720
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffca1d6c000-7ffcb0218000 [228.7M] r-x/r-x SM=COW  ...t_id=32b8255d

Thread 0::  Dispatch queue: com.apple.main-thread
0   libpolyclipping.22.dylib      	       0x10bedf9a0 ClipperLib::PointInPolygon(ClipperLib::IntPoint const&, std::__1::vector<ClipperLib::IntPoint, std::__1::allocator<ClipperLi

MariMakes, 2023-05-05:
Hey @mikerj1,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
Thanks for your report 👍

I can reproduce your crash on our side. I also noticed that a common workaround works in your case aswell. 
I am able to slice your model if you move it away from the center of the buildplate. I hope that helps you get unstuck. 
[image]

I'll have one of the developers take a look. 
Thanks 😊 These reports are super helpful in troubleshooting.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic