Ultimaker/Cura #15276: Slicing Failed

TypicPick6, 2023-04-18:
### Application Version

5.3.0

### Platform

Windows 10 Pro

### Printer

N/A - can't slice it

### Reproduction steps

1. Imported my file from my downloads folder, its a file from terrain tinker, seems like other files work fine
2. Hit slice

### Actual results

It take it a few seconds and the bar dose not go very far and then it just fails. I provided a screenshot, no error code given.
[image]

### Expected results

It should have sliced it so i can begin printing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11257201/cura.log)
[Test Scene - 0.zip](https://github.com/Ultimaker/Cura/files/11257205/Test.Scene.-.0.zip)
[Test Scene - 1.zip](https://github.com/Ultimaker/Cura/files/11257208/Test.Scene.-.1.zip)

TypicPick6, 2023-04-18:
The files where able to be sliced on other software and other files could be sliced on this current software. I have not modified or added outside scripts of any kind that would change cura, just added 2 printing profiles and something from the marketplace, settings guide.

fircop97, 2023-04-18:
I am having the same error: Failing to discretize parabola in 5.2.1 if I scale up above 175%. I wonder what happens to yours if you scale it down as a test. Could our issue be a build volume thing?

TypicPick6, 2023-04-18:
> I am having the same error: Failing to discretize parabola in 5.2.1 if I scale up above 175%. I wonder what happens to yours if you scale it down as a test. Could our issue be a build volume thing?

I kinda feel like its how much it needs to slice, since the file said it was gunna take over 3 days and another build i did was only gunna take a day and some change so it probably has to do with really long builds

MariMakes, 2023-05-05:
Hey @TypicPick6,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I took a look at your logs and I see this:
CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!
CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process
CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221225477

But I don't have the correct information to help you troubleshoot. 

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

TypicPick6, 2023-05-05:
I just reverted back one update and its working fine.

MariMakes, 2023-05-08:
Hey @TypicPick6,

Thanks for the update.
I'll close the issue. 

Happy printing. 👍

closed 2023-05-08 by MariMakes