Ultimaker/Cura #15277: Naming File Export - Select all type in box doesn't allow typing

EzPressPlus, 2023-04-18:
### Application Version

5.3.0

### Platform

MacOS

### Printer

Anycubic Kobra Maz

### Reproduction steps

1. Load part
2. Slice Part
3. When you want to change the name of the file saving to an sd card you go to the bottom left naming text box
4. If you click in that text box and select all the text and try typing it won't change anything but if you delete one character and then select all the text it allows you to delete all or type something new

### Actual results

Doesn't allow you to delete all text selected in bottom left text box

### Expected results

If you select all the text in the bottom left text box you should be able to delete the text selected

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

Vandresc, 2023-04-28:
Hi @EzPressPlus, thank you for your feedback!
I'll bring it up to the team to see how they can improve it. But I expect that it will not get a high priority to implement if it's not a quick fix. Fingers crossed 🤞

MariMakes, 2023-05-01:
Hey @EzPressPlus,

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15277
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15277

closed 2023-05-01 by MariMakes