Ultimaker/Cura #15278: Failed to slice

JoramNevversan, 2023-04-18:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

I upgraded from Cura 5.2.2 to 5.3.0 and it fails to slice any stl I load. All it says is "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker" and it sends me here.  I have tried uninstalling it and reinstalling it with restarts in-between and it still gives the error. I tried to see if it would do the same thing on my Surface Pro 2 and it works fine. The screen is just to small for me to use it regularly.

### Actual results

I reported the bug

### Expected results

Help with fixing the issue

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11257528/cura.log)
 It happens with any .STL I Try

JoramNevversan, 2023-04-28:
> Hi @JoramNevversan <https://github.com/JoramNevversan>, unfortunately, I
> cannot reproduce the issue you describe, could you please provide us with a
> project file so we can recreate the environment where you get the errors?
> Many thanks!

It does it on all stl files that i try to slice. I have tried sclicing
files that sliced fine in 5.2.2 and i have downloaded me versions of the
stl and got the same error. I have even tried it on new stl files i have
never downloaded before with the same results. I have since removed 5.3.0
and reinstalled 5.2.2 and it slices without any problem.

MariMakes, 2023-05-05:
Hey @JoramNevversan,

Welcome to the UltiMaker Cura Github 🚀 
@Vandresc is correct, we need more information from you to start troubleshooting.

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-05-29 by github-actions[bot]