Ultimaker/Cura #15279: z seam gaps

rafaelmontserrat, 2023-04-18:
### Application Version

5.3.0

### Platform

MacOS

### Printer

Sovol SV06 / Kingroon KP3S

### Reproduction steps

I've been printing lately cookie and clay cutters and since the moment I updated to 5.3.0 I can't get rid of these annoying gaps in my prints. They only seem to happen in certain orientation of the parts but I've had no luck tweaking settings to completely avoid them.

### Actual results

The results are shown in the pictures.
[image]
[image]

### Expected results

I am doing almost exclusively single walls prints so this issue ruins my parts. I've tried using outer wall wipe distance, not using coasting, deactivating linear advance, turning off retraction... nothing worked.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

rafaelmontserrat, 2023-04-19:
Hi, I'm attaching a project file where I had the exact same issue, I hope it helps 
https://we.tl/t-PSlVkQhNAW

GregValiant, 2023-04-19:
I use whatever partial roll I have laying around for these little test prints and today it happened to be PETG.  The changes I made to the file were to accommodate the material:  Hot End Temperature and Bed Temperature.  My printer is an Ender 3 Pro and so to accommodate the bowden setup I changed the Retract Distance to 6.0.

This is a shot from the back side showing the Z-Seam.  I don't see any gaps like you are showing in your prints.  
[image]

So I'm unable to reproduce the problem.  Coasting is off so that isn't it.  Have you checked your extruder to make sure everything is tight, and that there is sufficient spring pressure on the filament?  Your retraction distance of 0.6 tells me you have a direct drive.  Maybe there is an issue within the hot end?

rafaelmontserrat, 2023-04-19:
Hi. Thanks for checking on this. It is weird that the issue is repeating in both my printers and not in yours but it could also make sense since I do a fair amount of TPU and PLA and I change the extruder tension regularly. I do have direct drives in both of them. I would say that after your test my problem is not coming from CURA but from somewhere else in my printer so I will have a look on that. You can close the issue if you consider it finished :)

GregValiant, 2023-04-19:
I'm going to go ahead and close this.  We can always re-open it.  My thinking is that I just haven't seen this behavior either in my own prints or in any other bug reports (although yours could be just the first in what will be an avalanche...ya never know.)

lvanek23, 2023-05-08:
I am having this same issue! I was fine until i updated to 5.3.1

GregValiant, 2023-05-08:
@lvanek23 we will need a project file to re-open.

rafaelmontserrat, 2023-05-09:
Hi.

After this ticked was closed I kept having the issue but I just guessed it was my extruder, although it was quite well calibrated in my 2 machines. In order to keep printing those single walls items I started to use the options "outer wall wipe distance" and "retraction extra prime amount", being this last one the one that has given me the best results, although I currently use both. 

All that said, as soon as I go back to 0 on "retraction extra prime amount" my prints start failing again right away.

GregValiant, 2023-05-09:
@rafaelmontserrat  I was going to revisit this but the DownLoad Link has expired and I had deleted the original project file.  If you and  @lvanek23 could provide new project files I'll take another look.

rafaelmontserrat, 2023-05-09:
I will provide later today project files with my current working settings and with a couple things disabled that brings my prints to "old" (when i opened this ticket) quality.

lvanek23, 2023-05-09:
I re-ran e-step, resonance, and pressure advance calibration yesterday to rule those out, all were spot on needing no changes. 
  
  I woke up and decided to get froggy, turned retractions off altogether, still have a z seam gap. :( 
  
  
[image]

rafaelmontserrat, 2023-05-09:
This is from yesterday. Even with the outer wall wipe distance and retraction extra prime amount settings activated it managed to fail, so I guess is a good test. Attached an STL as well.

This profile has been working perfectly fine on my KP3S and I use a similar one on my SV06, just the extra prime amount is set to 0,216mm3 because I have a 0.6mm nozzle. As soon as I turn off that extra prime amount in both of my printers all my parts (single wall) start to show gaps in the Z seam. I've seem quite some people already complaining about this CURA version Z seam alignment and so on so something else might be happening regarding this... 

[image]

https://we.tl/t-6GmsltOfNr

LastDragon-ru, 2023-07-03:
I also have the same issue (petg, 0.6 nozzle&line, orbiter v2) in 5.3.1 and 5.4.0-beta.1 :(

GregValiant, 2023-07-03:
Without project files this is impossible to chase down.  I don't get the problem with my printer so I can't troubleshoot here.
Enough of you have posted pictures that there must be a problem.
Load the model.  Set Cura up to slice.  Use the "File | Save Project" command to create a 3mf file.  If you are using Windows then right click on the 3mf file in Explorer and select "Send To | Compressed (zipped) Folder" and post the zip folder here.

kastamauro, 2023-07-03:
[cilinder 5_3_0_alpha_xmas.zip](https://github.com/Ultimaker/Cura/files/11941842/cilinder.5_3_0_alpha_xmas.zip)
[cilinder_5_4_0_beta1.zip](https://github.com/Ultimaker/Cura/files/11941843/cilinder_5_4_0_beta1.zip)

The problem itself is lack of calibration, but slicer used to mitigate it optimizing wall order printing.

LastDragon-ru, 2023-07-04:
I've tried LA on/off, Outer Line Wipe Distance `0.0` and `0.3` mm (other settings: no retraction, petg, 0.6 mm nozzle&line, orbiter v2, marlin v2.1.2.1, delta). Result always almost the same :( 

[image]

[CFFFP_ZSeam (2x).zip](https://github.com/Ultimaker/Cura/files/11945856/CFFFP_ZSeam.2x.zip)

LastDragon-ru, 2023-07-04:
1 mm retraction and Outer Line Wipe Distance `0.0` gives me

[image]

GregValiant, 2023-07-04:
@LastDragon-ru Thanks for the file.
I sliced your file "as is" with the following exceptions.
I commented out the "linear advance" and "ABL" commands from the StartUp section of the Gcode file.  (Neither apply to my printer.)
I did not change to a 0.6 nozzle.  The 0.4 prints at 0.60 line width just fine.
I changed my "home offset" to accommodate your round build surface.
No retraction.  That string going across is pretty solid.  (I did remove the brim.)
This is looking at the inside of the Z seam on the near part (part 1).
[image]

This is the outside of the Z seam on part 2.
[image]

These are with my profile for my stock Ender 3 Pro running Marlin 1.8.  Retraction is enabled.
[image]

I just don't get those "Z-seam gaps" on my printer.  I print a lot of PETG.
A couple of questions.
You have "Linear Advance" enabled.  Are you sure it's dialed in?
What firmware are you running?  (I know the machine settings say Marlin but there are variants).

LastDragon-ru, 2023-07-04:
> You have "Linear Advance" enabled. Are you sure it's dialed in?

Thanks. Yes, with Linear Advance plugin + I see it in gcode, the value is `0.063`. Actually, the results are bit different when it on/off but gap is here.

> What firmware are you running? (I know the machine settings say Marlin but there are variants).

Just Marlin v[2.1.2.1](https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.2.1), you can find full configuration [here](https://github.com/LastDragon-ru/Marlin/tree/TevoLittleMonster-SKRv1.3-Orbiter-Mv2.1.2.1)

> I did not change to a 0.6 nozzle. The 0.4 prints at 0.60 line width just fine.

I've tried different orientations of the seam (with 0.6 nozzle), the gap is always here :( So I'm (re)calibrating my delta right now (seems most likely this is a hardware problem?) and will test again after. Probably I will try 0.4 nozzle too, but only as the last attempt (I need a lot of disassembling to set nozzle...)

GregValiant, 2023-07-04:
Oh, before I forget.
You have a G28 in your Ending Gcode.  That might be fine depending on how the printer Auto-Homes but dropping the Z with a part on the build plate is (in most cases) a really bad idea.  G28 X Y is an alternative.

GregValiant, 2023-07-04:
**_(seems most likely this is a hardware problem?)_**
I have my suspicions about that.  This problem has been reported several times (as you can see here) but I've not really been able to get a handle on it, and I have not been able to duplicate it with my printer.  I am meticulous about all aspects of the printer calibration though and a lot of poster's here are not near as fussy as I am.

I have an Excel macro that analyzes gcode files for "flow".  In this case, as in the other similar complaints, the flow is exactly right.  I haven't found an instance where Cura is shortening the amount of plastic required for any extrusion.  Even when Coasting turned on, the last extrusion before the coasting starts is always correct.
So yes, I'm leaning towards this being a hardware problem, but I just don't know for sure.  With all the little tools I've made up for analyzing this stuff I have not seen problems in the gcode that Cura generates.  That doesn't exactly let Cura off the hook, but it makes it unlikely that there is a problem in the gcode that _**I**_ generate.

GregValiant, 2023-07-04:
I wish I would have used a different color.  Taking photos of white is tough.
@kastamauro These two images are of a print using your 5.4 project file and I used 5.4beta as I did with @LastDragon-ru project.
[image]

[image]

I know its hard to see but you will just have to trust me that the Z seam is pretty much perfect.  It is in line, and everything is welded together as it should be.

I'm open to more suggestions but at this point I don't see a problem with the G-code.  I've analyzed it in Excel and printed models and everything looks good.

kastamauro, 2023-07-05:
@GregValiant Is the inside zseam looking good? mine is showing gaps due to optimizing wall order not optimizing, mine is oozing before printing the inside-cilinder outside-wall. I partially fix it using extra retraction prime but it looks like "wall order optimize" isnt working properly, i found another post refering to it. https://github.com/Ultimaker/Cura/issues/15056
Thanks in advance for aiming this issue.

rafaelmontserrat, 2023-07-10:
Now I can't find an explanation for this. In the same print, some of the parts are good, some others have gaps in the seam. Any ideas? Is it possible that my X and Y motor steps are off and thats why depending on orientation I'm getting this? 
I really don't know what else to tweak... Attached some pics and the 3mf file.
[issue.zip](https://github.com/Ultimaker/Cura/files/12005698/issue.zip)

GregValiant, 2023-07-10:
@kastamauro I haven't been able to duplicate this at all.  I don't print many single wall models but all my other prints come out fine.  I would characterize both the inside and outside Z seams (on that test print I posted) as "excellent".  They are straight, and everything bonds together correctly and there are no gaps. I haven't been able to think up another experiment to try to force this behavior.

@rafaelmontserrat It is very rare for anyone to have to adjust the X Y or Z steps on any printer.  If you are able to send M503 to the printer and view the response the steps/mm for all axes are in the M92 line.  It might be a good idea to check that to make sure they are at the defaults.  When the steps are off on the XYZ there is always a dimensional problem with prints.
I have a little app I wrote that will communicate with a printer running Marlin and I can post it if you want, or you can use PronterFace/PrintRun which also has a "Receive" window.

[image]

- That image _appears_ to show that on those top few layers, the outside feature extrusion coming from the left breaks it's path early and kind of moves in a tight radius to head for the inboard feature.  The printer should make it to the point it was told to go to.  If it is breaking off early, it must think it is at the point.
- The layers below the gap seem to have little blobs at the Z seam.  If there was some slop in the drive train then both the "little radius" problem and the "sat too long waiting to move" problem could be accounted for.  Now what exactly could be loose or slipping so that the print head doesn't move crisply enough you will have to try and figure out.
- The base of the outboard feature has the same problem as the upper layers.  The base is three extrusions wide and the top is a single extrusion.  That sure sounds mechanical.

I'm still leaning on this being mechanical problems.  Rafael's project file shows Accel at 1000 and Jerk at 5.  Those are not aggressive numbers so I wouldn't expect the printer to be over-taxed mechanically.  You could try dropping the Accel to 500 and see if it makes a difference.
If there is some play in the trolley wheels (not adjusted correctly to the slots) then something like what we are seeing can happen.  If you try to wobble the bed and the print head about any axis there should be no movement in any direction and yet both the print head and the bed should travel freely in their slots.

It still bothers me that this is happening on two different printers.  I can't check them but I'm pretty good at gcode and my analysis says there is no problem with the gcode and my printer has no problem printing your gcodes.

rafaelmontserrat, 2023-07-11:
Thanks for the input.

Lets say that this is not a gocde issue then, I guess.

You mention that there might be play in my trolley wheels, but my printer doesnt run on wheels, runs on linear rods and there is virtually zero play in any of the axis.

What you called a little radius after the extrusion breaks (like you called it) is oozing material after the toolhead did not make it to the end of the line. I was there checking the print and I could clearly see how the perimeter were not complete, the toolhead was moving to the next thing before finishing it, thus leaving the gaps. However, I'm still unable to tell why is it only happening when the print gets to be single walled. I added a chamfer on my design so the wall would be more robust and thats why there is just a tiny section of single wall, but still I can't avoid it designwise.

Also, the blobs in the Z seam that you mention are due to me forcing the printer to over extrude in there by reducing the pressure advance values from the optimal value that I found...

Thanks for your efforts in helping me solve this. Ultimately I am going to try different slicers to see if it makes any difference.

GregValiant, 2023-07-11:
Alright, some progress.
I printed a Benchy (because I don't have enough of them) and look what showed up on the Z seam...
[image]

I know my printer.  The chance of this being a mechanical problem is very low.  So this does seem to be a Cura problem.

@MariMakes here is a gcode of that print.  I did take a look at the flow settings and once again I don't see anything odd.  It doesn't print well though.
[3DBenchySpFish.zip](https://github.com/Ultimaker/Cura/files/12015144/3DBenchySpFish.zip)

MaddoxTheGamer, 2023-09-17:
Hey guys, im having the same issue

My printer is an ender 3 running Klipper and 0.4mm nozzle. It has an upgraded hot end and direct drive extruder. Im also using Cura 5.3.

I've been printing cylinders with 2 walls to test out different settings and try to solve this, and I found some interesting results.

This is what I started with:
[image]

Firstly, I discovered that lowering my speed from 120mm/s to 20mm/s resulted in a big reduction in the z seam gap. The seam came out almost perfect.
[image]
[image]

After this, I tested pressure advance at my normal speeds, and found that while it was off, the z seam gap mostly disappeared. I ran the tuning tower command on a 2 walled cylinder, and as you can see, more pressure advance caused more z seam gap.
[image]
[image]
[image]
The z seam still isn't perfect. I can also see the usual symptoms of too low pressure advance towards the bottom of the tower where there are no z seam gaps.

Im not really sure what to do now, as pressure advance is vital to maintaining good print quality at the speeds in which im trying to print. It's interesting that linear advance did not have the same affect for you guys. Maybe the firmware has something to do with it (Marlin vs Klipper)? 

 Hopefully we can solve this!

Here are the settings that didn't make much difference:
-Combing
-Retract at layer change
-Wall ordering
-Input shaper (resonance compensation)
-Retraction (on or off)

GregValiant, 2023-09-18:
Part of this may related to Klipper.  There is some work being done by the Cura team to better integrate Cura and Klipper but after reading a few of the bug reports regarding blobbing and Klipper it appears to be an involved process.

geonosys01, 2023-11-01:
Came across this issue because I was struggling with a similar one. One thing though I noticed. I can see the gap in the Cura preview (5.4.0). 

[image]
[image]

Trying to see if there is a setting that fixes things, but I thought I'd mention it here.
[project.zip](https://github.com/Ultimaker/Cura/files/13225313/project.zip)

GregValiant, 2023-11-01:
@geonosys01 you are severely under-extruding.  Here is a close up of your model.  Those gaps shouldn't be there.

[image]

I would suggest you calibrate the E-steps on the printer and also check your hot end for a blockage at the bottom of the bowden tube (where it meets the nozzle) or maybe a partially clogged nozzle.
I would also suggest you slow down your retraction and prime speeds to 35.  You are printing kind of cool at 200 and the plastic is pretty thick.  Trying to snap it back that fast might have something to do with the under-extrusion if the thick plastic can't react that fast.

Here is your model printed with silky gold (that's been laying around for a while) on my Ender 3 Pro.  I was too lazy to change nozzles so this is with a 0.4 at .2 layer height and .4 line width.
[image]

geonosys01, 2023-11-01:
Thanks for taking the time to respond (and print).

My shrink also says I have other problems :)
But seriously, the thing is I also printed this three at once. interesting to see they don't all have identical issues.

[image]
[image]

I agree there is an underextrusion issue I need to address, but I've been printing lots and lots of things (with same filament, same settings) and only these are giving me these issues.
But since you can print this file without issues, I agree. There must be more so I won't litter this thread any longer.

GregValiant, 2023-11-02:
_"...but I've been printing lots and lots of things..."_
How is your preventative maintenance schedule?
About every 40 print hours, I pull the hot end apart, clean it all out, trim the bowden tube back about 5 or 6mm, clean out the heat break, pull all the trolley wheels and clean them up and lube the bearings with some light oil, make sure all the axes are still squared up, check for loose screws, check all the fans for strings they might have picked up, etc, etc.
These machines work hard for hour after hour.  They need a little love once in a while.

I've been the greeter here for a couple of years now.  I have a few tools I use to analyze gcode files and this sort of problem with the gap at the Z-seam has not (to date) been in any gcode I've looked at.

geonosys01, 2023-11-02:
> _"...but I've been printing lots and lots of things..."_ How is your preventative maintenance schedule? About every 40 print _"They need a little love once in a while."_

Don't worry, they're getting enough TLC :)
Took this one apart before starting these print runs and also yesterday because of adding a filament sensor. Tonight I'll re-calibrate it and see what happens then.

_"I've been the greeter here for a couple of years now. I have a few tools I use to analyze gcode files and this sort of problem with the gap at the Z-seam has not (to date) been in any gcode I've looked at."_

I'm curious about the tools you use. Are these publicly available or is it something you put together yourself?

geonosys01, 2023-11-03:
Ok, re-calibrated the lot and still had the issue. Upon further investigation I cam across the arachne engine and more people having similar issues.
So I set up an older Cura version (4.13.1) and the results are interesting...

[image]
[image]
[image]

Now the results are as I would expect them.

GregValiant, 2023-11-03:
Your results are better.  The print I did was with 5.5.0 so Arachne was enabled.  I haven't had any problem with it myself.  It's different mostly because there are additional settings to pay attention to.

geonosys01, 2023-11-03:
Hm.
Settings are the same for the printer, filament is the same, settings for the slicer are the same. Only the version is different (and other engine).
So what would be the additional setting according to you?

GregValiant, 2023-11-03:
Other than changing the nozzle size (too lazy to put in a 0.6) I sliced mine with your settings.  Line width and layer height were different, but they were not far off from my own.  I printed it at 75mm/sec so the flow rate was about 6.3mm³/sec.
The main difference in the 5.x versions of Cura have to do with the line width as a consequence the settings in the "Wall Section" need to be understood.

Did you happen to try slicing that in PrusaSlicer?  It has a switch and you can turn Arachne off.  That allows you to check both ways.

Flopalop2, 2024-01-28:
I removed my gaps by calibrating and upping Outer Wall Wipe Distance to a larger value like 1-2mm. Tuning linear/pressure advance, flow, and esteps is important too. It mostly was a problem for me on curved surfaces with a travel move directly after the seam. Those travels happen more newer Cura versions due to the new variable line width simplifying inner walls. Now, on single walled objects, wipe goes in the wrong direction. I am reporting that bug today.

yhd343762, 2024-05-20:
[image]
hi, i have the same problem, i mesured the gap is arround 1.5mm.  I copy the GCode into Rhino,in the picture pink is start path, red is back path, green is the travel path. i fund the extrude did not go back close the start point, and the distance between end and start point is 1.477mm, i think it is the reason of gap at z seam. but unfortunatly i dont know how to change start end point distance to make them closer.

yhd343762, 2024-05-20:
this is My GCode, i found the extruder did not work at the first point "G0 F18000 X164.61 Y193.023 Z1.4",which close to End point(X165,Y698,189.905).
G0 F18000 X164.61 Y193.023 Z1.4
;TYPE:WALL-OUTER
G1 F600 Z1
G1 F2700 E355.77066
G1 F3600 X163.134 Y192.962 E356.01633
G1 X161.65 Y192.824 E356.26419

GregValiant, 2024-05-20:
@yhd343762 please zip and attach the full gcode file so I can analyze it.  What Cura version did you use?

The "G0 F18000 X164.61 Y193.023 Z1.4" line is simply a travel move going up to the start of the next layer (plus your Z-hop height of 0.40).  I'll need to see what is happening prior to that.

yhd343762, 2024-05-20:
hi，
I use cura 5.6. This is my gcode file, it may be different, because I
change some parameter to try to fix it, but the problem is not fixed.
I am a Robot 3d Print engineer, normally we will add a compensation for
start end position, if a curve is closed.

yhd343762, 2024-05-20:
Thanks for your email：）

GregValiant, 2024-05-21:
Github rules are that you cannot post a file here via Email.  The file must be posted directly to the website.

yhd343762, 2024-05-21:
> Github rules are that you cannot post a file here via Email. The file must be posted directly to the website.
Hi Greg,
this is my printed part Fotos and GCode. thanks for your reply.
best regards
Haidong
[ppp.zip](https://github.com/Ultimaker/Cura/files/15386021/ppp.zip)

GregValiant, 2024-05-21:
Excellent model that is easy to analyze.  I picked the seam location at gcode Layer 5.
Reading the Gcode into both Excel and AutoCAD, each layer above 4 is almost identical.  The nozzle comes into the Z seam, backs up 0.10mm (wipe distance) and then travels to the other wall.  Each of those travel moves between the walls is an **Un-Retracted** move.
This is a zoom of your image.
[image]

The key word here is "Un-Retracted".  There should be a solid string between the walls which would form a vertical surface connecting the two walls because there is no retraction, and there should be oozing.  We don't see that.

This is a zoom shot from your gcode file when viewed in Cura.  The 0.10 lines are the wipes and they are in the opposite directions that the associated extrusions were.
[image]

On every layer, the last extrusion of a wall meets up exactly, to three decimal places, with the first extrusion of the same wall.
This is the final travel move of LAYER:4 to the starting point of the first extrusion of the inside wall on LAYER:5.
G0 X147.004 Y148.496
and this is the end point of the last extrusion of the same wall.
G1 X147.004 Y148.496 E983.35293
They match exactly.  There is no gap.

In the Cura Simulation view you can check that by moving to the area with the horizontal slider, then click on the slider "ball", then use the left and right arrow keys to step through the gcode.
The AutoCad analysis shows the exact same thing.  No gap.

Whatever is going on with these "Z-gap" reports I just cannot find anything in the gcode files to account for it.  I think there is something going on with either the printers or maybe the filaments being used.

It is apparent to me that the Cura generated gcode is correct.  I've printed a couple of the gcodes that have been submitted here and I can't duplicate the problem with my older Ender 3 Pro.  I'm going to remove the bug label from this as it must be a problem with the printers, or with the material because it isn't the gcode.
I'm going to close this because I think we've beat the subject to death.

In addition @yhd343762 is this line:
**;ELEGOO Cura 5.6.0 - 231229**
which would read:
**;Generated with Cura_SteamEngine 5.6.0**
if it was sliced with UltiMaker Cura which is the app that this site supports.  Elegoo has their own fork of Cura and you should make a report on their site.

closed 2024-05-21 by GregValiant

yhd343762, 2024-05-21:
@GregValiant Finally I find the reason.every time at the start position the extruder move on it but didn’t extrude anything. So I find there is a function named „Extra length on restart“, I change it from 0 to 1mm, now the wall are connected, but it extrude too much. I will reduce it later.
[image]

yhd343762, 2024-05-21:
I have solve this issue, 2 methods:
1 complete close retraction
2 use orca slicer, they have a function „ scarf joint seam“ .
I think method 2 is better. Because some times I need retraction, and it makes seam almost invisible 
[image]
[image]

belbiy, 2024-08-16:
.6 nozzle and Cura 5.6.0 also giving me huge gaps in seams like that. Switching to OrcaSlicer solved the problem. Unfortunately, i like Cura interface better, but for .6 nozzle i simply cannot make it work.

jonathankoren, 2025-04-17:
I agree. OrcaSlicer fixed this Sauer with me.

I don’t have the gcode anymore, but I watched the model print. The nozzle literally never moved over to close the gap. The size of the gap also seemed to correlated to the layer height. (Doesn’t make sense to me, but that’s what I saw.)

GregValiant, 2025-04-17:
Cura now has "Scarf Seam" in the "Experimental" section.

There were a few reports of this behavior (open Z-seams) for previous Cura versions, but I have never had it happen in one of my prints.  (Most of my prints are PETG with a 0.6 nozzle.)

There was also the thing that the gap was not in the Gcode and when I printed the gcode from the original project, there were no gaps.
This appeared to be a machine problem rather than a bug in Cura.

Nobody likes buggy software.  I don't like it, you don't like it, the developers really hate it because their name is on it.