Ultimaker/Cura #15280: z-hop only on 1st layer

lgstern, 2023-04-18:
### Is your feature request related to a problem?

When printing TPU the nozzle leaves a trail between models on the bed. It is not a problem after the 1st layer.

### Describe the solution you'd like

Being able to enable z-hop just for the 1st layer

### Describe alternatives you've considered

Extra retraction.
Could clog an all metal hot end. Also with TPU it might not happen fast enough

### Affected users and/or printers

Those printing multiple models with TPU

lgstern, 2023-04-20:
> Here. You get to be a lab rat.
> Unzip the folder and copy "AlterZhops.py" to "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\scripts"
>
> It will be available in Cura under "Extensions/Post Processors/Modifiy Gcode" and then select it from the list.
> Set the New Zhop Height to "0" and the layer range from "2" to "-1" (-1 means the whole file). It will then set all the Z-Hops above the first layer to zero hop height and negate their effect.
> AlterZhops.zip <https://github.com/Ultimaker/Cura/files/11286709/AlterZhops.zip>
> I have a pull request in for this but there has been no action on it yet.

Thank you!!

sleewok, 2023-08-09:
> Here. You get to be a lab rat. Unzip the folder and copy "AlterZhops.py" to "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\scripts"
>
> It will be available in Cura [...]

Can you clarify the settings a bit? How do these settings relate to the setting in Cura print settings? 

With the following settings would z-hop be applied to only the first 3 layers?
Example setup:
- Z-hop off in print settings
- Plugin z-hop height set to: 0.3mm
- from start of layer: 1
- to end of layer: 3

Thanks!

GregValiant, 2023-08-09:
The plugin sets the Z hop height to what you enter in the plugin setting and for the layer range you select.  So you must have Z-hops enabled in Cura and at the height you want.  It doesn't add code or delete code, just changes the hop height.
The plugin settings would be:
Z-Hop Height:  0
From start of layer: 4
to End Layer:  -1

Layers 1, 2, and 3 would have z-hops and starting with layer 4 the rest of the hops in the file would be set to "0".

sleewok, 2023-08-10:
> The plugin sets the Z hop height to what you enter in the plugin setting and for the layer range you select. So you must have Z-hops enabled in Cura and at the height you want. It doesn't add code [...]

Thanks for clarifying. I tested and it works perfectly. Great work!

Malte-G, 2024-03-28:
Thanks for this Plugin its really annoying to have Z Hop turned on throughout an entire print when you only need it for an inlay for the first 2-3 layers.

GregValiant, 2024-03-28:
It was this model that got me to thinking.
[image]
When the horizontal hole was closing up - the feather edges warped upward and caused the nozzle to really smack into them.  I needed z-hop at 1mm to clear the warped edges.  Z-hop wasn't required above or below and the script came into being.
And now you know the rest of the story.

Cysign, 2024-08-21:
Wow, this could help a lot. Is this also compatible with newer versions?
Any chance we can ask Ultimaker to make it a stock feature of Cura? Is there a feature-request + vote system available?

GregValiant, 2024-08-21:
Yes, it works with everything from 4.13.1 up to 5.8.0.  There isn't anything "version specific" in it.
The Cura team was busy with other things and new post-processors have not been a priority.  I've got a lot like 'Alter Z-hops' that sort of take care of niche situations.  I keep my Github page up to date [HERE](https://github.com/GregValiant/GV-PostProcessors).

GregValiant, 2024-09-15:
Since you folks have seemed interested I have put together another Z-hop script "Z-hops for Travel Moves".

[image]

It is independent of the Cura settings so you can have Z-hops on or off in Cura.
It is a slower running script since it must check the length of every travel move in a file to determine if there should be a hop.
Looking at the settings in the screenshot:  the script is active, it will act on the initial layer through the top layer and any travel over 10mm will invoke a 0.50 Z-hop.  The script is on my Git page [HERE](https://github.com/GregValiant/GV-PostProcessors/blob/StartPoint/scripts/ZhopOnTravel.py).
At some point I might try to combine the two scripts but it might turn out to be an "either, or" thing.

rwetzel67, 2025-04-22:
> Here. You get to be a lab rat. Unzip the folder and copy "AlterZhops.py" to "C:\Program Files\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\scripts"
>
> It will be available in Cura [...]

Thanks!!! Very useful for inlays / 2 color prints!
I am a happy lab rat now.

GregValiant, 2025-04-22:
I decided not to submit "AlterZHops" and went with "ZHopOnTravel" instead.

The difference between the two is that AlterZhops acted on Zhops that were inserted by Cura.  ZHopOnTravel does not require that Z-hops are enabled in Cura.  It just checks the travel distance and if it is greater than what you enter then a Zhop is inserted, (and a retraction when necessary).
With AlterZHops you would enable them in Cura and then set them to 0.0 height from layer 3 up.  With ZHopOnTravel you would have them start on Layer 1 and end on Layer 3 and the rest of the file would not have Zhops.

It sounds the same but pellet fed extruders can't retract and so could never use Zhops but they will be able to with the new script.

closed 2025-04-22 by GregValiant