Ultimaker/Cura #15284: The wall leyers are seperated. Inner wall of outer wall

RadoBGMTB, 2023-04-18:
### Application Version

5.3

### Platform

windows 10

### Printer

Ender 3 V2

### Reproduction steps

Any model I slice

### Actual results

very weak bonding betwean outer and inner walls

### Expected results

solid print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

PK-        ×[×ÿÿÿÿÿÿÿÿ  3D/3dmodel.model  ÊP     ‚c      ­½ËÎ&GŽ%¸Îê„XÍ,ôëó»{BªÞô¦7Ì% 
U©¡KBŠLTÏÓyÌŽÝü·ãîª	 *á‡tºI£‘4û¾ýoÿùËÏ_ýóóïüôÛ¯ß}Þ^¾úüë§ß~øé×ÿîÃ?¾üøõþá¿ýÛ¿þË·¿üöÃçŸ¿2ð¯|÷á?¾|ùû_¿ùæOÿñù—¼ýòÓ§ßûã·¿¼}úí—o¦~ùøë?~üøéË?~7.ß|úí÷ÏßŒ¯aùæ5~øê¿þôå»¿üôóÏ?ýòùËçß?8Ï¿þüÑ_÷ù×¯ÿöã~ýã¯¿<|Í/ÛOÎ¯‚°ýûC>ÿý·þñé‹
Gä´R¢ïrúþóÇ_~þøåËOŸðý¾?õtÿø92±AÜ(ÏŸ
ôÇçOÿøýó§ß~ýòù×/þuÇ7¯ùƒÍî_¾µ¹øøÃÇ/¿úõã/Ÿ¿ûðÿüôåçÏ¾úò¿þnù?ÿû·_ýý÷Ï|þýŸöÃ‡ûñó¯?|þý«¿þðÕÏ?ýjÿõÏqüörKƒÿöß?}þÃÿfoøë§ß~þí÷ÿý·üý«Ÿ~øîÃøáßŒ¢ø×€ûíûÿùùÓ †QœÿëóÇß¿Š¯üêþwÿ?JÅ4éþú·¿ýÿþÝ‡‡×ký<_oË§õëùûÏË×ßžæ¯Ç—áÓ~Ì?Ó÷ødéóÿ—Ú›	øDaÿÿáó~õŸß}˜æým›§i5Ûø_ß}°ÿÿÿ~÷áëéõ6¯ýµøê›È$py—hxÍo/ü¹I½•¯ü¯QßX¿r¼øÜŠš¯¼ Z†·uXÃü‚
ìZÐ`Ï||MÝß_£§ååã¦ßöeÞö£Gûz;¦ezÿ½6Qó¸Î&Óûï}½mÃ±Îï¾wyg{óÒ§]ö}›Ì¥œe^ßæu]SúÞ{SÂ÷eÞÂXõtqy½´qr†{DµÖÑö—]ŒŽ¿éL4ÌÓ2ëO;ë0^|S$æùm,ìËå¼¢Þ—c7‘l&NÔëÛò2ý°‘êÍÅ6LëÚ¡6-8^Æ¼O}ŒËgòônÓ¿}[·©Gmz?Ú°˜lïI>¾¯qVÔ&Ø6v¨ó:¯mëêïøöZÖ©Gýz[_ûèÚýþ¨o¦ÞÛ´¿ýîãm^SÿÕ“Íï¸¸:¼óÙûÛ¼ŒËÜ¯émß—e}Ÿx{›§Ãž÷ÄžßV“«ClV»mÓÜUíÅ$›÷Îl-oËz¬¯î€­oÓ4õÞ<›†¯£«'ÛÛ¸==™ÜS×AVÄ6¬Xñ.h\Ú}>ÂôFr0;QÃ³Î0r›ÕHã£òzÍ]Ghó±¼æ),‘Æçøò9z_õì{_Ã8T²Ù¼ûäêØ£-Fª½’)Úhž¬»(¶ ½¨¥Q¶ÃTdÜY¼ÿ3ªmëqÌc¯Ë6ug·Z9ã‹n,™…ñæhL½|}vXÛþ`Ö›<‡…D¯ÕþØG='Þ,,(}õ£7ëÛàRÛT?ó0ZÔöDîÚáµÔ×¯p—­àˆÛ!{DÜNÖ#âVM®ˆ«å©%¾^Šµ­ûq;`ˆÛ©zD|Ò±+jFs8ÐŽØU4P·#öˆ¸±GÄíˆ=">Ø5u~µ#v}Äíˆ="nGìq;bˆO#vM½äp·±ëh· nGìq;bˆÛ{D|±+ê:Xm<ÿU°Z7r?¢mìm3Sh¹¢­ƒóf°®ƒó"²oëm3Xh›ÁzDÛÖ5m¹iëj3RídšÁzDÛÖ#Úf°Ñ6ƒuE[o¾šÁºÚ|UÄÍ`=¢mëm3Xh›Áº¢­7›Í`]m6+âf°Ñ6ƒõˆ¶¬G´Í`]Ñúvuž¸­oË7×¯iéîAx±„²Í`ùž×võÝ½µÓ®¯¸mËi÷½¿íOÍ`v»{S§Ý˜´jËi·©›™ÛsG¬¸Ý\BF¸û¾—í¯×=ŽQ¦Yæcìî¹ßží1¹šh^ó&²$ƒííSb4ÒÌoûzìý9ð}úÁTT¢YÇi˜ìoïïÓ§·éuXösžhìß¶½«'–wXWÛÝ—4ž{±€_ä›,Übâ&¾gz›yï'{æ·‚¦M­]WBž‰µ´Õ¨´Ä³™žjÎÄz>ªy?k¨­%^¬ 4®Òì3ñ T¼6¥wˆeÑ£~³ULFûcI"O…˜=ª[ °ÇÉ2³ý1K|LÛøGï]}o·ïžü~üÞÝ’7þçÏÐ6>ú‘ÌöFZûóg^lÉÍ7Kzo^j¸óÅµb¶3¬õ²¢=ÍðÚÓ? =ÍðÚÓ? =ÍðÚó?!>Íðq±^
õ_­[•Ïkføm3Ãh›~DÛÌð#Úf†Ñ¶3üŒ¸™á+âzYkgX¯jíi†Ðžføíi†Ðžføíi†Ðžgø	ñi†5qõ´3¬ƒžŠö4ÃhO3ü€ö4ÃhO3ü€ö4ÃhÏ3ü„ø4ÃWÄEäÞxé«¾
¨›~DÛÌð#Úf†Ñ63üˆ¶™áG´í?#nfø‚xŸÞ†yK
D^×ò?ˆÒ†ýmŽ£Û ²ÍVÓµ0ÖÐV·^kZ¬·mèÕ{kZ“¢|ïm-sC«ežKâZ?@µ­ÎÌÖ“ôþž·¡i6+WÄ¯Ñz¬PÉ&§øÐÉ&ç5YóCçÅF»[×Eì1Úe™ç1Äß“µúŒö·Î Ï/ËF¯ÃîÏmrŒÖ>ãåœ,ü6ÚÅ‘¬íñÝM¾Ó®–Ø‰ÅxW(Ë"dÚ}´ÔŽIñ^‚Ài-
ð¸½íÇ>Lö&¯m)×á
ïÓÚ7YÏeÞÞ,0­ñ{—·iØÑÒ£µ>¸…cµ•Åu{ïb©&ŸôíºÍì×·u´LK0£]­ß¬ÿ½–4²ä{­.?¾oõß»£µdôß»6XQ7ìMÙøì½Ûb½‚¶|¯ìfíVq~öâ½å÷š6lÓá}CAæ«ï-Çy~8Îåüš–Y_ŒwJ…÷êù­õÊÃ‡a_­Ö«ZŸ'k±4Žsdaì…>—v4Úl¿ŽlƒÚŽjûµ v^ ¡þ½Wö[û
£-tãÚoX÷&ncýÞÉ’dfÚ=g÷²žšÙV÷õ÷šO1ÓïÛ‚µ=lìÌiëq¶€ýeÝI=[ðnÔàm9¿&…µ¹½o¿¶ôÌãD™k½²q´5Êg¬G;[›bLNZB¸hÊ± Å˜îz´–+E÷³Ë\Û‘eKÌz{ýóËÒ8ã2ð½µýÚ¸ÛêÙµý—=7¶9ô÷Ö~Ãh³+›¿žÌå{kuýÞò{k?yý½å8×þùzœËù­×…ëù-õª^®ôªÖçz¼ÒçÚŽêõ÷ÊŽ< Êö[¯ûWökV·šrDÝhãa°%´«–#ßÄ®Wí{­Nd]Š=½²šìæ‹è{ñÆd]u¶ÄtiÍ.aÕï­ã×d
³]Û÷¤ÿ¾¸\N[Ï¯gÞ­¹ëëÌ÷oÞÎGÚ2ÞðÉ,©kƒ–<5Ïà+nxoé7¼´Xßdo¬Ð¨Âª[o˜Lö5ý9²8è°îóèŸkûÍ‡ÎÖ~Ü¯uŒZU.È\û
£µ°§¿.Ôï­ýÕõ{mÒ÷Ö~òú{Ëq®ýóÕ8×ó[¯Wó[ëUo\ëU©Ïõ:h©©Ïµµë¯¶£Ú~ÛuÿÊ~K¿ÑÆÚozÙÙŒÜÄö‡õCwma´˜ÿe-ÓA'ëïußnÚÞÕçÁ:Ú×ƒ¶_³­GÇ(ö)ÖÉ;›µÄ÷Öók	f³@.ß_Cm]°£\»k½²>^ÛN÷×_;]h-Ø± ÜÄ&Ó°Kw¬^oX8Æª¶£ám²2swkñ†u6³m1C¹_°ï9V«õõ¾×h_Vx¾®öF»îæmñÞÚ_]¿·üÞ:Þ¸úÞzœëxãjœëù­ãëù-õª^®õªÔç6ÞÐú\ÛQo\ÙQm¿õº¯íwš½“öìvÈÕßþ`S÷µ
»ªä±5dò™ÛBoíw‰7ÛnÄsdƒoÔW;îdmÛ«o¶“6XÞmùvbë8±ÔÎ]b3aŠ°ì†mÕ-–7¸óf3âÅöH6Bx³­%Ûvxö<[žAÈ
âé°%þ…áÅ›m¥±Ãƒ7¿ÙˆgKDi‡l´sú·¾yÚýÔY ~YÚãž’ñlg?c»ˆm°}³wç›Ý MÁ8`‰N/š½Gl®ñT°c2#¼I¼úh³KÉ•d´šÇfwÞl¹ëŠ¨7*‰¥þ°ñ¾Glóº{B'Û˜ë¿÷æÅòwià†a;ëÛ£½¼íæbPìÄ‡7rÝœ*ËZ
é<ª;ƒyÚ¼¹!¶hö®U™œ•ØÖŽ¹[þÏ
Øfãg°{>Ì>²š*Su?‘v›¸RK}Ïv÷”¤RÏÃÃ%ËJÜS¡Ò0Ì7X®ãî<7&i]uñôÛ±gp¼Ù¦%ˆ{Ä•2bËU{:úqã É¡»Ä•ëµ*Ä¶Â¥Ýxól=K¥Ó?¬…ñØnúm#®–›ÃÂÏ
aâ7[ T.t–M°øÓ÷Ë÷ˆ«%v7^žÁ¿K\-îv’y‘¡¿÷æ*¬°úÎhç˜ï¹!ÐXœð$ ©ß|M,¾ùšXŒö5±˜ç+b©a×ÄB·/ˆÛ€¦¶ªkâ* ©íùŠ¸	hjOrM\4F\ø°+bé=¯‰…ß¾"nšzÅ¸&kÕ5±X%¯ˆ›€¦^Ÿ¯‰EdpM,b’ÇÄe4tM\‰]Ça×Ä"¼"nÏ*æØóš¸RKpQï5qÐÔñöqÐX­ˆô¯‰ÅãŠ¸	hl_Uìn®‰«€Æˆ‹}ÕqÐÔ;ºkâ* ©÷’Äm@c)Ñb{M\4õþùŠ¸	hêû5qµÄ¼È\W‹{­¸&®š:OrM\ehª†ž%h,äù;ìr/
+¿ØòDžž¹M[õà5Ümú×2—ÁŒå=9soßÝDËVWöÜŒR¹ó½u˜þz³óM–Ú¾EÛD2^þ“™Ïñ–œ¾÷Ú*Œ1RÛøñš_[geŒÔ½ÖÍ·Ö!Œ³=%sóµµ£~Ùöy8ü´Ò‘ë Æ.ñ„ÌM™ë„Œ‡¼óáWkÝ{o¾˜²%Ñ‹whëèÅ’Ïž¹·‡jÒ– Ž§êî½·Œ],	ì¹˜{?{oIëh£uÝýÞ*rñæªÁ·îÉ\. x"æÞ®¯IÄØº`y˜Øæs•K6Ú2l1wli˜ÁNÃÝ‘¹ÖIË¶¶U½I[Û‚º%a¼iäÎ{k4-³ÌpÏ9×¦ï]Á¶}ñâþ÷VÇi-sWŸë€Åh-s7KWç¾-ágù—»¹ˆÚ±ÛÊíé—{ãÜ,(VD÷ìËÍ¥V,”öâß½q¶,F¹€Z$í¹—{¶`´åÂíë—0Ý˜ß¶¢c-œ–y¹çÛÛÌK¨\9U„T½÷š¶ÿ½×´ýq¾¦-•z~¯hë¤K­W×´eÎ¥ÖçÚ&P©íèš¶Ì¸Ôö{E«üÆ5m™o©ýÕ­ò“×´e¶¥öÏW´j]¸¦í¯G×´e¬R¯ƒW´u¬R¯¿×´ýuÿš¶o<£­ãœkÚRæ:¾º¦-c•:®»¢­s,u<yM[êFÇ^Ó––:~¾¢­c•:n¿¦-ó+õ~áŠ¶J¯ÔÛ”kÒ"¹RoŽ®H«HÅC¼%»&-+õNð‚¶É«Ô;ÐkÚ2­Rï|¯hë@¥Þq_Ó–h½Ó¿¦-î:ÃpM[ufãŠö½«°¯î´SlÅýÙ¶m2Ïhcäyuu±|¡¾öØ¢j+ÊÚŸlZAxál§hlKäíïvÅ®v©™%Ía'K¼2Ü£±+€y÷u|]Ñ²X¦¨ûë·H8v&ÛûXosï=~ÑÙÌ6HcÕL»Z¹{*Èî6_­q1v½¿CÚú^º§r¬Ëc·½®o†Ò,.¯u×õÆÀBª×l]9ö†Lc}®ÖïÑ}Ïl›ê=u]EÙ,³|“×…ßk;©f§OØÁiÌ-Û4w;³í~ñ}±=t9§‹…b³ípÃ»ïY,;¹oh
Jßc‡í^ß¥±çË<ð6è Ûj‹¢µ£w;¡íhçaWÓÆN÷Hc= ¾)ìÁjµs«BÇÎÍHcŠmpü4Å»²mÞ½ÞÜæ½ZÀm7œwuô]k»˜¨Ûei²Ù¡­z¬W[ê·ÅYvd³1°º8›Éâ÷Xo§]ÓÑ»ÚúÃ-!c¹œrNWûÆÅŽöÞ³˜nYV£ÒÑÕôÚl¡«o¦;f§µ±w›uàzäwÇÚtôe·¹ÄÖ¥ð=‹Í™„ëvÓÏÖ½8YÏŒ¡“¾-~óùËvÆÍRÓË® ¯hÌ'–?±{W6³í—]^Û‚éŽ½§K3Zm×^Æf¿ø=6nË€Â»ï±Þ\Û¬7²Y
ÂN"yçÅû46Öƒ,çÔüÿ´í}½6o×l±¯.ÈæG­íôP÷´…_f¿¼"¯%Ö§Ý‘m²ó
'ûñkzTÅ»4¶J.Šh²¨5T¿Ñù9‘¿º`'cÎ_ôgh®*ö»q¬òVWÉÈzÕ3ôµ—êÅ÷Ä_F¨¬ák;ü¢ŽKh~¤9}õÆZÊ–øá«_ÖeSSÿ5j³Ø'ŸÝü¢Éu=ÒÖÜX…W’kê§ï~6jí»kê‡ïn~§ââ»íèXio¶<[Š[âGkrKühqn‰­Ò-ñ£åº%~´n·Äð†øÙJÞ?ZÒ[âGk{Küh‘o‰­ö-ñ£e¿!¾Zÿëx³¾‚óYÜÙÐ>Òì†ö‘b7´ôº¡}¤Ö5í3­nh)uCûH§ÚG*ÝÐ>Òè†V+ôd×Âlv
5žp³«]ì2=6hÛýƒvØÞ*&ïÇÀNk?`3’Öî^°³°±tl´ëdÇ>zÁÌfñè¼Zã>ö»Ÿ—JçŒöðzŸ µ˜5ŠÖ
m¼èp°s…›]¡Ó»ÖÞ;Vïõ\r)¬Ôf´v?*öïÅüök<Fë›ð^Ë~M›ï¤µËwŽÅ
o]Zßn'ZÛ*á<Sx¯_úäw¶ôhm÷ÂëiwÛ¥ÛQçX,oa×Kâ½¶µáþÛh×ÝR0!J5ZKNØgõßk7Ürã´vjÔw±.³8­ÐØÝƒ¸nØf•sd´–pÉ´¶Yûcå9?OÃq.nâ°÷ÚO*Ùý™í
)ëÕç7—9ªv‘]j]wœv^­'êd-³åŒ¬1¿·Ç·“ þ<<?ÔŒ•±;+º¿^f´–¬b>ª™#ß‘ØùÕžn­uãÀbm/n´¥nX¾m±3õ6Ûï¯]ÜéÛ|³³@[ê¤I´K-z'c'{“¥Ý˜Û²¿•¶`% Sè¾Ì–sÞ,5mÁïÏÈ6h´–Âèåyü½&Vš#Û2¶Òlìýïµ7Yâ"îÙŸc´–ÎìÝ¶09ÚzL²¿*}ÑÚ‰ÜþX™g4#¥-Ô>ÖæÈ.2èþHÝfóï‡ÃâXíbýÑýîôÍŽEwæ×i'»Õ;Î¯yÕÁÎ~FÛ·œ‡yîÞm)–Çð“'¼zÛ¾¾´#+`{:²§WFk©-Ä3ÚÉ%Ú¾åµìüäÖ»ºÚh§m‚‡rôÛxªl°r§IÔ½ÆÛhívmÞîdš²­ïåÝ_ÙÖÁº±Ö®¯³<ŒõâÅ¿Ñz“L<;h´v?¹¾Þ8[Lqxr’v´Ná|}x¯e½peûû:Yù+Ëð•ã|á¯ÌÇ~²¡½ð“µ6ZË¿Ø±XúvíŸ-ÞÌë‚ÓÚz·Ø;ë‚­ûy=rÚÁÂ„¨“×ëQ±­]›d¿Ç÷^­ƒc^vòÅ,Ú‘5Ÿ_¬¿Åºo´Ö^ÇÛ»,f¸\÷s¼a´õðÄàx#Ç9Fû²T`<ô|#Î)â+ÏìÛ
ÖŒ®â«*®3ÚÉÖjÎïU\WÅ“–}¶ÛíÂ¦0Gñ¤Y{á'Ú?i´…nd¾öÏÅºÐŒÕÕºP­GÍ­µèöòè¾–ë`£F{±ë¯Ñ–:i´ëo±îmiWë~omiƒWñFçmiû×qN_9más®ã«"®sÚÂ×]Äu¶¦ë‚¯ˆñþ³u2ô[°½ZêµÌhí·­­ž´WkY±†­õóò ê5´X»ÍüGƒ¸?º^»‹˜ÁhËµì:f(b£µj«iÃ÷^Ç*EŒä~ÃNÌÆóàö½:Fª×ÐÕ¢û×…ë5´\Mû‹½Æ#ZÛYþyZoôñ?a¬®ßk¥¨[|µÚaû]J®ûæEü¬»ñ¤×>­Él÷¼Y™6¼÷šöð°Þ¤|ï½öó¾{ÏnëQ¦=½WÓZ¼áwOvÞkÕ:ÛK÷âºŠöôÞ+Zû%ÆüÍ8û DµïÆ„ö¦‚¶yï
ZKÂÄ=ìé½¶´Zÿ{-7BÚÓ{/i-òŽ½§÷Ú9Û¤Š÷fÚÓ{¯h­oƒ?/Ô¼×rnVÂíÝêèùœ‚¶yï5íË®‰{ºÓ{Ë©õnÑò÷´§÷JZÛïù­¶ý‹ýBC[ûœG´ö½¥¿zF[ûØ+Z‹	—ÔãÔŒ•Ia—–úI³wíÈãÉLÛÌÑ5íb¿[{]Nïµ‹Ìú{g{oA{zïí`{ÌOžßk¿µã«qï{Úó{%­­ô~#wÈœÞkù»îµ÷ÞŠöôÞ+Zo»ŠùÓ{­SÊ.¢í¿· =½÷ŠÖ!Ü{ÄÞgë²‡ý÷´§÷jZo NW¾wµÔu?oVÑžÞ{Ek©"äÔßû^»$Ü.°î}¯õŠeÚÓ{%ífmõve3×£*¿a÷Û‘¾n¿Œ]ñí7'›w1CKë\ºÿ}×ŒÖ’Þvìï=Z÷íËÒëtÿl·u³çÆâçÂ×ùÃ–·z¶`´ö˜7ªŸh-d	õžÌöÃ‘øí÷÷e¶›Ë‘±xÿ{­±Á‚³Ø+uzïn¿9ßÏ×YÝÃÚäâíÆ'ZË0ÚqÂžÌžK0óíŒ³•ƒº=ˆæ]}þ»ãl·ã äûßkÙksåúÛÎ‘Ç|ýußv&ÖðúþXY>Ç~¡£Ÿc´‰ùþ˜ëkÆ
wZýº•ß˜~2°¡µµÊ<l?k¶ï·‚½;Î¶;´Ë¡úúlûî1dè<ö®ÇÊ2C–Oî®)›Õ5¬>öþüÚXXR­×/jÅ1«FXKmXSš÷ßÍ›"{ze»¸—©Æ»ßë=qVÇêÒZ]Ã~c3æùë÷Z
Èö¨bœ-'c…ÍwÇÙëGûÑ7,§j@Þzz¯·ýöëevŸC®ãœi­½×}èû¶à·Ðn¼üDëEÊ~,ê?Gb1…åI7ü§;¬Üà{¶÷ßk:kYoÿ¦÷hí@/ê=ÚÅÒó±Æw’Ù{ØÕ{mmàOŠÔ´Ö/iö)ÆÙö©¦´ïÛ¾é¤]¥×ÿùË¤Ø¹ÌØk~z¯/²ÒL¬^=Ý°>~SW+¦õiûï5ûµã­~èº7ÎýïõêŠpW¯Ì¯tÇÙž]Œsw~ý¾bÛ…îâ½]½2Z+éØîª÷½BŸ-ÞðÄûïvd´Öï‹^‡÷ÇYØ¯Ç9VÑêÆÏÊo8­Ô
‹G,yßO^¿÷°R‰{œFŸ¯¾×{!:~ãrœ-Ÿg?Ñó¾
^Í¯ç -8w]¸Ò+ïß™;¶¥ÏOZ%õ]_g:©íÈjp¼ëëÌöµým®-6stå7Ô{m}•þJ}ï•ŸTã|åŸÅüÚz¤×¡WF«×#¡Ï—ë å'{vd´zýök´V‘ë~«œsA2V©h½«¢È!_Å9­íÿ4­ï´þô{Oy$ù½Õ>´~¯ûØ‹}h¹‡­¾÷m=ÎÏhëù}FkÝ¥E]ã’¶ŽcO´Ö£íÅ:XÚQK«×ßÚ~[Z½î×~£¦}¶µ´:Î©ýdK«ã«Ú?·´OÖ£šöÙzÔÒê8¶^jÚgëQK«ãv5ÎÏÖ£ê½æŸŸ¬G-í“õ¨¥Õû²z=jiŸ¬G-­^„ßðµìÁþ÷Ù{Íûß–Vo½_hiõ8×ûß–VÏ¯ÕŠýoK«õªÞ—U´÷¿-­¶£zÿÛÒêxÒbäbÿ[Ó^Å±ê½WþªÞï×ï½ŠÛÕ8_íÄü>Üÿ¶2?ÙÿÖ´Ïö¿-­^ëýoKû`ÿ{ŽÍîÇgÚûñÆ‰VooœhÄ
í£xãDû Þ8Ñ>ˆ7N´â†öÊ~«ýï‰öA¼ÑÐ>Š7N´ÚOVñÆ‰Vç'«ýoMû,Þ8ÑêõHè³½W¯ƒU¼qz¯^«ýï‰öA¼q¦½oœiå¾»Š7N´ú{«xãD«Ç¹Š7N´z~«õèDû Þ¨iŸÅ'ZmGÕº¢}o4´â­Î×UñFCû(Þ8Ñjÿ,æ÷Y¼qzïƒx£¡}oœhõú[Å'ÚñF¿òÜˆŒ7ª÷žie¼QùíÕ÷–uº6¥ãÊo4ï½Š7*½:Ñêx£ŽŸk™¯ãç²nÕÒêx£òÌWñF½O©ßûÈo4ï½Š7*¿q¢½ò“e®–ù*¿!æ÷2Þze´z=ªÖÁú{/ã
[... cut ...]
¾CØ»ÝÔ‚ˆ½xñ*}n°ù!IEßÞ-((wföcæpzØQÜ)¦Á;e!A3¾\§`Êíz§ãjy¸Ò¨3gR?„$øÉ%}Îa˜LOV§Ârì´>ZùŒmnº#ÜH¹ÅøÍ ¦.~°¢~úƒÚÁDŸ|›ã-VÕnjµÉSäŠÌ-+”å¬û†FµŽeX?bñ¶.® çAø³ˆ•PK- -        ×[×ÿÿÿÿÿÿÿÿ        €   ÿÿÿÿ3D/3dmodel.model  ÊP     ‚c              PK- -        T.Óÿÿÿÿÿÿÿÿ        €   ÿÿÿÿ[Content_Types].xml  T      î       Äc      PK- -        sÖã–ÿÿÿÿÿÿÿÿ        €   ÿÿÿÿ_rels/.rels        ³       ÷d      PK,       - -                             çe      PK    óf         PK        ÿÿÿÿ

GregValiant, 2023-04-18:
Thanks for the report.
It would seem likely that vÑnPøÊ„6}í¢]—k´éU��z¹¸&‰¹ Ãæ��Áh+ì0�-‡�XVË|mvl\£E­Œ¥²)3šK�¢·“u2Ê�½®¨¬°TÂª×­q XVÈ¢,�‰úØÌþË��²˜#�Ö¸�VI�"S›��ÐŠ^ˆƒ!jBÏÂ‡f4óš³på�šá�èá8�¢‚%–puGeFÓ¨D@{ø3×Ð�—Xyh-2•�š®C�´�ZeÀ; but without a Project File there isn't any way to tell for sure.

RadoBGMTB, 2023-04-18:
I couldn't send a project file. This is a .3m file opened with notepad. I
couldn't send Stl or 3m.

MariMakes, 2023-05-05:
Hey @RadoBGMTB,

@GregValiant is correct we need some more information for troubleshooting.

Have you tried reverting to the default settings?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.