Ultimaker/Cura #15285: Ability to blast fan on the support-side of a support interface

trussell051693, 2023-04-18:
### Is your feature request related to a problem?

When doing single head printing, it is frustrating sometimes how hard it is to remove the supports.

### Describe the solution you'd like

I really like the idea behind the "Supported Skin Fan Speed" setting, but I'd like to have the additional capability to blast fan on the support-side of the model-support interface if possible, as opposed to being restricted to blasting fan on just the model-side of the interface. Blasting fan on the model-side interfacial layer can make that layer so weakly bonded to the model material that it might come off with the support material when you are trying to remove the supports. Obviously, that is bad.

### Describe alternatives you've considered

I've been working with ABS and if I'm going to be able to get the supports off reasonably easily, I seem to have to resort to not using support roofs or use them with low density and/or an undesirable amount of separation distance between overhanging model material and support material. Doing so can lead to very bad surface quality on the model-side of the supported interface though.

I'm dreaming here, but another idea I had was what if the support settings were as comprehensive as the model settings, but distinct from the model settings? For example, if you could print your support material at a colder temperature than your model material, or with more fan, etc. you might be able to make the supports substantially weaker and easier to break away when you are doing single head printing. I know that we already have that capability for doing dual printing, and people who have dual head printers can accomplish exactly that - but, to my knowledge, single head printers can't do things this way.

### Affected users and/or printers

Users who have single head printers or users who want to do single head printing on a multi-head printer

GregValiant, 2023-04-18:
Unzip the folder and put the "AddCoolingProfile.py" file in the "....\UltiMaker Cura 5.3.0\share\cura\plugins\PostProcessingPlugin\ **scripts** " folder.  It's a post-processor and will be available through "Extensions/Post Process/Modifiy Gcode".  I had a pull request in to make this part of Cura.  I'm not sure what happened with that.
[AddCoolingProfile.zip](https://github.com/Ultimaker/Cura/files/11264481/AddCoolingProfile.zip)

trussell322, 2023-04-19:
Hi GregValiant, thanks for passing that on so promptly! Having a feature like that is a good idea I think. However, I do have a couple of comments. It blasts fan on an entire layer without distinguishing between the support and the model material. It is also not a general solution like the current Supported Skin Fan Speed setting in the sense that it does not apply fan to the appropriate spots automatically. Instead, you have to check out the slice and figure out which layer(s) you want it to blast fan on. I was hoping that if someone had already figured out how to create the Supported Skin Fan Speed setting, it might not be such a big step to figure out how to apply fan in the same way, but on the support-side as opposed to the model-side.

GregValiant, 2023-04-19:
Have you used the "By Feature" settings?  That's what it's for.  Separate fan speeds for Skin, Outer-Walls, Support, Support-Interface, etc.

trussell322, 2023-04-19:
Got it! I didn't see that drop down. This is a cool extension. My gcode preview stopped rendering properly soon after I got the extension though - my support material and brim are now invisible. But I opened up the gcode in another slicer and it looks like everything is ok. Perhaps I have a graphics card issue or something and that's why Cura is having trouble displaying the slice.

closed 2023-04-19 by trussell322